gnuton/asuswrt-merlin.ng #117: Wired Access point

drabishek22, 2021-09-07:
**Firmware Version Affected **
<html>
<body>

Firmware Version:386.03_2-gnuton1  

</body>
</html>

**Is this bug present in upstream Merlin releases too?**
Yes 

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
 I am using the DSL-Ac68U as an access point without connecting to a DSL connection. internet is conected via LAN port. The VPN does not connect neither does NTP server uipdate time. But all other websites can  be accessed via WIFI

**Screenshots**
If applicable, add screenshots to help explain your problem or remove this section

drabishek22, 2021-09-08:
Can i request that a an access point feature be enabled in the firmware, I managed to fix this by setting up ethernet lan to wan option but the network is set in another subet and cannot communicate with the primary router

gnuton, 2021-09-08:
can you please link here the issue in the upstream merling github?

gnuton, 2021-09-20:
since this was a issue with the config I think we can clos this.

closed 2021-09-20 by gnuton