gnuton/asuswrt-merlin.ng #118: Open NAT Game Profile's not saving

chalkybriman, 2021-09-15:
**Firmware Version Affected **
386.03_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
Not sure

**Describe the bug**
On the OpenNAT tab on the left under general the game profile does not want to save and apply.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to Open NAT
2. Click on Enable Port Forwarding > Add Profile > Select Game > Select Internal IP > Click OK > Apply

**Expected behaviour**
I expect game profiles to be listed for my internal IP

**Screenshots**
Not Necessary

gnuton, 2021-09-15:
hi, DSL-AX82U or DSL-AC68u?

gnuton, 2021-09-15:
just tested on DSL-AX82U and it works.

chalkybriman, 2021-09-15:
DSL-AC68U

gnuton, 2021-09-17:
Bug confirmed. Affecting only DSL-AC68U.

chalkybriman, 2021-09-18:
No worries thanks for confirming I'm not going mad :D

gnuton, 2021-09-18:
Fixed. I am building a new stable with ROG and Open NAT enabled.
The new stable with the fix will be available in a few hours: please update
to 386.03_2-gnuton2 once available.

closed 2021-09-18 by gnuton

chalkybriman, 2021-09-18:
awesome thanks

chalkybriman, 2021-09-18:
Sorry I have that version already few hours yet before a new release?

gnuton, 2021-09-18:
I have not published it yet. I am testing it once more before release..

gnuton, 2021-09-18:
released 386.03_2-gnuton2.
I think you had 386.03_2-gnuton1

chalkybriman, 2021-09-18:
confirmed working now

gnuton, 2021-09-18:
Eccellent! Thabks a lot for reporting!;)