gnuton/asuswrt-merlin.ng #119: OPEN VPN client is not starting after restart

KarolKosMBAVI, 2021-09-30:
**Firmware Version Affected **
Stable: 386.03_2-gnuton2

**Is this bug present in upstream Merlin releases too?**
I don't know.

**Describe the bug**
I have a VPN Server - OpenVPN serwer 1 configured. It is working fine but till the router restart. When the router needs to be restarted,   using Reboot Scheduler or after a power failure, an OpenVPN won't startup. It hangs on the step: Export OpenVPN configuration file. It saying to wait when the file will be ready, but it never is. The way to run OpenVPN again is to turn it OFF and again ON manually.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the VPN section.
2. Configure an OpenVPN server 1 using basic settings and run it.
3. Check if you are able to connect to using OPEN software.
4. Reboot ASUS DSL-AC68U.
5. Go to the VPN section. -> VPN server. see the message in "Export OpenVPN configuration file"

**Expected behavior**
Once configured OpenVPN server should run with no issues after every reboot of the router.

gnuton, 2021-09-30:
Is it the same issue in this thread? this https://www.snbforums.com/threads/stuck-on-export-openvpn-configuration-file.29103/page-2

KarolKosMBAVI, 2021-09-30:
yes, it looks like the same
do you want me to attach some logs? when to capture it?
can I help somehow else?

Have a good day.
Karol.

 
From: Antonio Aloisio
Date: 2021-09-30 10:31
To: gnuton/asuswrt-merlin.ng
CC: KarolKosMBAVI; Author
Subject: Re: [gnuton/asuswrt-merlin.ng] OPEN VPN client is not starting after restart (#119)
Is it the same issue in this thread? this https://www.snbforums.com/threads/stuck-on-export-openvpn-configuration-file.29103/page-2

gnuton, 2021-09-30:
This seems to be a known bug. try to ask in the forum first...
I guess the solutiuon will be something in the config after the reset of the router

viciovb, 2022-01-25:
Just installed 386.04 on ASUS DSL-AC68U and I have the same issue. My OpenVPN Client didn't start automatically, no problem with the server.

gnuton, 2022-01-25:
@viciovb have you tried to follow the instructions in the forum link?

viciovb, 2022-01-26:
@gnuton to be honest it's not a big deal for me, I don't reboot my router so often so I can wait for a fix but I'll take a look at the workaround thanks