gnuton/asuswrt-merlin.ng #120: unable to use usb-hdd

nztianmi, 2021-10-08:
hi great work as i can use my dsl-ac68u as a repeater!
unfortunately my Seagate USB-hdd stopped working. front usb indicator doesn't light up, hdd powers up though. admin page shows no USB device connected.
any suggestion?
thanks a lot!

gnuton, 2021-10-08:
Hi,
does the HDD work on another computer?

nztianmi, 2021-10-08:
> Hi, does the HDD work on another computer?
> […](#)

yes works with latest asus firmware, and works on xiaomi router 3 as well

gnuton, 2021-11-01:
Hi,
The errors in the logs are these lines
```
1:17:35 kernel: usb 2-1: new high speed USB device using ehci_hcd and address 6
Oct 22 21:17:35 kernel: usb 2-1: device descriptor read/64, error -71
Oct 22 21:17:36 kernel: usb 2-1: device descriptor read/64, error -71
Oct 22 21:17:36 kernel: usb 2-1: new high speed USB device using ehci_hcd and address 7
Oct 22 21:17:36 kernel: usb 2-1: device descriptor read/64, error -71
Oct 22 21:17:36 kernel: usb 2-1: device descriptor read/64, error -71
Oct 22 21:17:36 kernel: usb 2-1: new high speed USB device using ehci_hcd and address 8
Oct 22 21:17:37 kernel: usb 2-1: device not accepting address 8, error -71
Oct 22 21:17:37 kernel: usb 2-1: new high speed USB device using ehci_hcd and address 9
Oct 22 21:17:37 kernel: usb 2-1: device not accepting address 9, error -71
Oct 22 21:17:37 kernel: hub 2-0:1.0: unable to enumerate USB device on port 1
Oct 22 21:17:38 kernel: usb 3-1: new full speed USB device using ohci_hcd and address 6
Oct 22 21:17:38 kernel: usb 3-1: device descriptor read/64, error -62
Oct 22 21:17:38 kernel: usb 3-1: device descriptor read/64, error -62
Oct 22 21:17:39 kernel: usb 3-1: new full speed USB device using ohci_hcd and address 7
Oct 22 21:17:39 kernel: usb 3-1: device descriptor read/64, error -62
Oct 22 21:17:39 kernel: usb 3-1: device descriptor read/64, error -62
Oct 22 21:17:39 kernel: usb 3-1: new full speed USB device using ohci_hcd and address 8
Oct 22 21:17:40 kernel: usb 3-1: device not accepting address 8, error -62
Oct 22 21:17:40 kernel: usb 3-1: new full speed USB device using ohci_hcd and address 9
Oct 22 21:17:40 kernel: usb 3-1: device not accepting address 9, error -62
Oct 22 21:17:40 kernel: hub 3-0:1.0: unable to enumerate USB device on port 1
```
So the problem as I thought is something related with power.. For some reasons there is no enough current coming out from your USB port.
you can workaround this using a powered usb hub.
If this is a sw and not an hardware issue... this may be something that could be have been fixed by asus within the new GPLs.
For this you have to wait for the next major release.. hopefully coming in about one month...

closed 2022-03-08 by gnuton