gnuton/asuswrt-merlin.ng #125: SSH Certificate error

GitKobo, 2021-10-31:
**Firmware Version Affected **
386.03_2-gnuton4

**Is this bug present in upstream Merlin releases too?**
Do not know

**Describe the bug**
If using or trying to use or produce a certificate for router with certificate option in administration and system tab, there are always default IP in certificate. Can this be corrected any way?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Administration'
2. Click on 'System tab'
3. Scroll down to 'Service' and export Certificate
4. Expand Cert.tar and See error in IP and other details if Router has set to use any other IP than default 192.168.1.1

**Expected behavior**
Exported certificate should use modified IP and other settings from DSL setup

**Screenshots**
[image]

GitKobo, 2021-10-31:
My mistage, after 3rd factory reset, changing ip, creating new certificate in WAN with using DDNS tab certificate shows finally correct ip, I must have done some mistages earlier.

closed 2021-10-31 by GitKobo

gnuton, 2021-11-01:
no worries! :D