gnuton/asuswrt-merlin.ng #127: RT-AX82U: get_cfg_clientlist is undefined

gnuton, 2021-11-16:
This is currently the major problem on RT-AX82U which doesn't allow the router to turns the UI to not working after reboot.

gnuton, 2021-11-16:
Analysis of that issue:

1.  var current_node_count = [][0].length - 1;
2.  he code that generate that is "var current_node_count = [<% get_cfg_clientlist(); %>][0].length - 1;"
3.  so the problem is that get_cfg_clientlist somehow doesn't return anything... that's a web hook
4.  cd /www && /usr/sbin/httpd -i br0 -p 5556 will run a new instance of http server
5. The problem seems to come from 
   shm_client_tbl_id = shmget((key_t)KEY_SHM_CFG, sizeof(CM_CLIENT_TABLE), 0666|IPC_CREAT);
   https://github.com/gnuton/asuswrt-merlin.ng/blob/gnuton-master/release/src/router/httpd/web.c#L26518
6.  adding debugging, I noticed errono is 22 which means the shmget fails because the key is already there.
7. KEY_SHM_CFG in hex is 0x7d1
8. we can now run ipcs to get the mem segment id:
   ```
   # ipcs
   key        shmid      owner      perms      bytes      nattch     status       
   0x00006666 0          gnuton     666        2568       1                       
   0x00007777 32769      gnuton     666        6404       1                       
   0x000003e9 65538      gnuton     666        172160     1                       
   0x00008887 98307      gnuton     666        5892       1                       
   0x00008888 131076     gnuton     666        9476       1                       
   0x000007d1 163845     gnuton     666        9932       1   ```       
9. to understand the process who has created the mem segment we run
```
    # ipcs -pm
    shmid      owner      cpid       lpid       
    0          gnuton     1214       1318      
    32769      gnuton     1223       1318      
    65538      gnuton     1292       1336      
    98307      gnuton     1316       1318      
    131076     gnuton     1316       1318      
    163845     gnuton     1336       1457 
    ```

gnuton, 2021-11-16:
replacing RT-AX82U with RT-AX58U router/cfg_mnt/prebuild have fixed the problem.
AiMesh page seems to appear, get_cfg_* works correctly now

washran, 2021-11-16:
@gnuton The particular file you got for the rt-ax82u seemed to have gotten corrupted somewhere. It happens unfortunately and you have to investigate. I have been building asuswrt-merlin's test firmwares since 2012. I like the docker setup better.

closed 2021-11-17 by gnuton