gnuton/asuswrt-merlin.ng #129: Modem Asus AC68U - Wireguard VPN- Richiesta di supporto

silvermindphasar, 2021-11-21:
**Wireguard VPN**

Sarebbe una buona idea aggiungere Wireguard al nostro ac68u con modem incluso, sia lato cliente e server insieme a openvpn da
menu VPN del modem stesso.

gnuton, 2021-11-21:
Afauk wireguard requires kernel support and i am not sure DSl ac68U
supports that. Asus will be adding that to specific routers. No idea about
this. I can ask.
Once it s added i can evaluate the port.

silvermindphasar, 2021-11-21:
Thanks for the reply, I'll leave you a site where you can implement wireguard on an rt-ac68u router. 
https://nixfaq.org/2020/07/wireguard-on-asus-wifi-router-with-broadcom-armv7-32-bit-chipset-running-asuswrt-merlin.html

gnuton, 2021-11-22:
i have just seen wireguard is coming to merlin
https://github.com/RMerl/asuswrt-merlin.ng/commit/be6bce8fae407fdb5c33e98b9d60ef9e57d900c2