gnuton/asuswrt-merlin.ng #130: LAN ethernet connection not working

damocle81, 2021-11-26:
**Firmware Version Affected **
386.03_2-gnuton5

**Describe the bug**
I have an alarm system connected directly to my DSL-AC68U using the lan port, the alarm is able to connect to the router all is working but after some hours the alarm seems connected (blue led on the router is on) but I can't connect to the alarm from local network and from external network, using the web gui of the router and the asus app the alarm seems connected. In the alarm system the gateway is 192.168.0.0 (the ip of the router) and the IP configured in the alarm is 192.168.0.90. I tried with no success to change the alarm IP.
To reconnect the alarm I have to restart it. This problem started last week when I installed the last firmware 386.03_2-gnuton5, I reset the router after the installation and applied the saved configuration.

Please could you help me to solve this problem?

Thank you

gnuton, 2021-11-26:
Hi,
To me this sounds like an alarm problem.

Which version were you using before 386.03_2-gnuton5?
If you were running 386.03_2-gnuton4, I have to say it's exactly the same
firmware since all changes in gnuton5 were only for DSL-AX82U.

To troubleshoot the problem try to use different ports (it could be an
hardware issue too) but if the router says it's connected.. then it's
connected! :D
Another thing you can do is to ping alarm from the router to see if the
link between the two devices is up.

I do have plan to release an alpha soon with up-to-date sw. you may think
to test it if you like or just switch back to the previous version of the
fw if you think that would solve the problem.
Keep me posted.

damocle81, 2021-11-26:
Hi,

thank you for your reply.

I don't remember if the previous version installed was 386.03_2-gnuton4  or 386.03_2-gnuton2, but only after the last firmware update I performed a reset of the router, maybe this is the difference?
Regarding the LAN port I tried all the LAN ports.
I will try the next release.

Thank you

damocle81, 2021-11-29:
update;

I forgot to insert in my first message one important thing, the DSL-AC68U is configured as router in a AiMesh network, DSL-AC68U + 3 ZenWiFi AX Mini  XD4, could AiMesh be the cause of the problem? I found a thread with similar problem with wired client and AiMesh: https://www.snbforums.com/threads/rt-ax88u-messed-up-dhcp-on-wired-devices-beta-1-and-3.68836/
Have I to use official firmware instead of gnuton firmware?

Thank you

gnuton, 2021-11-29:
AiMesh that we run on Merlin is the same as the one in the stock firmware.
The only difference could be the version of the aimesh/firmware itself.
In your case it seems to me the problem is in something related to the
configuration, for some reasons if i got it right
you cannot connect from router-> to alarm because of some routing issues.
A factory reset and a re-config may help. If you do not have other special
setting, you can of course try to get back to stock.

damocle81, 2021-12-01:
@gnuton thanks for your reply

As soon as possible I'll try with ASUS firmware with factory reset.
Regarding the static IP, the alarm is requesting a static IP, I can't leave blank the requested IP in the alarm configuration.
The connection is off also if I modified the router configuration to assign the same IP to the alarm.
If I re-start the alarm the connection is OK, but after some minutes the connection is off, I checked the router logs but I can't find logs related to the alarm MAC address or IP, before activating the AiMesh nodes (3 AX Mini XD4) the connection between alarm e router was OK.

damocle81, 2021-12-06:
Hello,
I tried the official Asus firmware, same problem, no problem for the wired device to enstablish the wired connection to the router but after some minutes the device is disconnected. No special settings, the device is requesting static IP, no static IP configuration in the router. I will contact the Asus support.
Thank you

gnuton, 2021-12-06:
are you sure the problem is in the router itself? try to set a computer to
get a static IP from the router and check if that works.
99.999 % chance it will. :D honestly If I would start to debug this issue..
I would first check that the problem is not in  the alarm itself.

gnuton, 2021-12-06:
in any case you can also try to not set static IP to  alarm at all..
and you will see the alarm getting a dynamic IP and then losing the
connection for its own reasons...
this is another test you can do.

damocle81, 2021-12-06:
Hi, the alarm can't be configured with dynamic IP, only static IP, I'll try to assign static IP to a computer wired connected.
I'll let you know
Thank you.

damocle81, 2021-12-06:
Hello,
I tried with a notebook with linux wired connected to the router, the connection is OK, with dynamic IP the router assigned an IP (192.168.0.49) to the notebook, if I set a static IP (192.168.0.150) in the notebook the router assign the same IP assigned the first time (192.168.0.49), in the GUI page the IP is 192.168.0.49 but if I check using the notebook the IP is 192.168.0.150, in fact I can't ping 192.168.0.49 from the notebook or from the function in the router GUI and I can ping 192.168.0.150. It seems that there is at least a cache problem about the assigned IP. I leave the notebook wired connected, I'll check the connection.

gnuton, 2021-12-07:
Static IPs usually use lower numbers than dynamic.
Eg 192.168.0.2can be one, this in order not to clash with dynamic addresses.

damocle81, 2021-12-13:
Hello,
I tried with other modem/router (Netgear), same problem, so the problem is the alarm or the ethernet cable.
We can close this
Thank you

closed 2021-12-13 by damocle81

gnuton, 2021-12-13:
glad you got it. :D 
The problem is in the alarm. I think it gets stuck or something. not a cable issue.