gnuton/asuswrt-merlin.ng #131: DEV_45958_3 RT-AX82U

washran, 2021-11-26:
**Firmware Version Affected **
386.RT-AX82U-45958_3

**Is this bug present in upstream Merlin releases too?**
Merlin not supporting RT-AX82U

**Describe the bug**
A clear and concise description of what the bug is.
1. client list showing wired clients only.
2. General, Aimesh empty.
3. Diversion module Lan, Diversion not showing anything after 4.2.1 upgrade.
4. Cannot set ip on inbound firewall rules.
5. System log showing kernel: CPU: 1 PID: 17073 Comm: dcd Tainted: P           O    4.1.52 #6 frequently.
6. Bandwidth monitor showing wired clients

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem or remove this section

gnuton, 2021-11-26:
Please do not use 45958 because it's not for RT-AX82U yet.

gnuton, 2021-12-21:
fixed mergiing hte latest RT-AX82U GPL.

closed 2021-12-21 by gnuton