gnuton/asuswrt-merlin.ng #132: Unable to upload firmware onto Router

lamham25, 2021-11-27:
**Firmware Version Affected **
386.03_2-gnuton5

**Is this bug present in upstream Merlin releases too?**
Yes/No

**Describe the bug**
I have an ASUS DSL-AC68U router and typically download the DSL.trx file to upload manually onto my router. For some reason, when I upload the firmware onto my router, it takes a while to load and refreshes the page back to the login page.

Is this to do with the file or my laptop?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Asus Login > Firmware Update'
2. Click on 'Upload '
3. Wait for it to load
4. Refreshes page back to Asus Login

**Expected behavior**
The new firmware uploading onto the router before refreshing to apply new changes

gnuton, 2021-11-27:
Hi,
that could happen when you are not reallylogged in the page. I mean you open your browser and the router's page and navigate to firmware upload page, but  all data comes from the browser's cache.
Then you upload but actually you do not have the rights.. so what happens is the page refreshes.
This can be once scenario, but there  could be others too.
I would suggest you to try with a  browser in incognito mode and try to be in the same network.
let me know how it goes

gnuton, 2021-11-28:
@lamham25 have you managed to upload th new firmware?

lamham25, 2021-11-28:
hello gnuton, yes the new firmware uploaded, thank you!

I just need to figure out why my personal laptop is stopping me from uploading the firmware.

gnuton, 2021-11-28:
ok thank for the update!

closed 2021-12-28 by gnuton