gnuton/asuswrt-merlin.ng #136: VDSL2+ Setting & SNR Adjustmen not showing

JustifYI, 2021-12-17:
**Firmware Version Affected **
386.03_2-gnuton5

**Is this bug present in upstream Merlin releases too?**
Don't know

**Describe the bug**
VDSL advanced settings are "hidden", and I can't set VDSL SNR adjustment with negative values as I could on Asuswrt firmware before.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to http://192.168.50.1/Advanced_ADSL_Content.asp

**Expected behavior**
See all the advanced settings and be able to set **negative** SNR adjustment on VDSL2+ again, since I can't reach the DataRate speed as before.

**Screenshots**
Now: [image]
Before: 
[image]
SNR Adj. now: 
[image]
SNR Adj before (ex.): 
[image]
DataRate before: 
[image]
DataRate now: 
[image]

gnuton, 2021-12-18:
working on this.

gnuton, 2021-12-31:
SNR Adj has been fixed and it's in the latest published beta.

BTW from where this comes from? is this a DSL-AX82 config page?
It's a mix of VDSL and ADSL options...
https://user-images.githubusercontent.com/47337816/146549215-084599f5-c77f-45ea-b922-2c744db6e848.jpg

JustifYI, 2021-12-31:
The screen is just an example since I don't have access to the original page, but yes, a lot of features like "UBPO", "ESNP" are gone now from my "DSL Setting" page.

Actually, the most important setting for me is the "Stability Adjustment" (VDSL2), since I could set it to -2 and gain about 15.000 Kbps in Downstream Data Rate.. but now I can only set positive values only (with the actual firmware: 386.03_2-gnuton5)

I'm gonna try your latest release now :)

And yes, my modem is a "DSL-AX82U"

JustifYI, 2021-12-31:
Okay, now I can set negative values on "Stability Adjustment", but the SNR target remains at 6.2dB, even if I try to manually set the SNR via command. Do you have any advice?

[Can set negative]
[image]

[Target still at 6dB]
[image]

[Tryied with cmds]
[image]

gnuton, 2022-01-02:
let me take a better look at this and I will come back.

gnuton, 2022-01-02:
Hi,
I had a quick look at this today and my understanding is that it would work in case the DSLAM would not ignore it.
I can be wrong, but at moment that was my understanding and what I have seen.
in the firmware I have not found anything wrong.
There is chance also to play with 3rd-party dsl drivers too.. but I am not sure how that would work.
The driver is in /etc/adsl/adsl_phy.bin
You could google for alternative ones in case they have some "improved" capabilities

JustifYI, 2022-01-02:
I've tried to find something new for the driver, but I think that mine is the latest version(?), actually running "A2pv6L046w.d27m". (But to be honest, I don't know so much about it)

The sad thing is that I could adjust the SNR with Asus stock firmware, so I'll try the following:

1) Bring back the stock firmware and see if the SNR adjustment is working.
2) If it does then I'll put back your latest release and try again to adjust the SNR.

I'll let you know. Thank you for your support btw! :)

gnuton, 2022-01-02:
yes please let me know if you see differences between my builds and the
stock.
If you spot anything different I will come back to you and fix it.
Thanks for your help

JustifYI, 2022-01-03:
Okay, with the stock firmware the SNR Adjustment is working. I've tried with -3dB and +10dB. 
Then I put back your latest build and sadly the feature was not working anymore.

Stock FW used: DSL-AX82U_3.0.0.4_386_45660-g37ee6d6
Custom FW used: DSL-AX82U_386.04_0-gnuton1_beta3

[-3dB stock fw]
[image]

[+10dB stock fw]
[image]

[-3dB custom fw]
[image]

gnuton, 2022-01-03:
different driver version. It was missing in the GPL that i got from Asus
then i used the one in my previous version.
I will release a new image with new driver and let's see if that fixes it.
I will ping you once available.

JustifYI, 2022-01-03:
Should it work if I replace the adsl_phy.bin taking it from the stock FW and placing it in the custom FW?

gnuton, 2022-01-03:
you must rebuild the firmware for that to work.

winter2222, 2022-01-03:
I also noticed that DSL driver in your latest build is quite old, nice to see that you are gonna update it soon.

JustifYI, 2022-01-03:
By the way, I've noticed that even in the stock FW the following settings are not showing anymore:

Rx AGC GAIN Adjustment
UPBO - Upstream Power Back Off
ESNP - Enhanced Sudden Noise Protection
G.vector (G.993.5)
Non-standard G.vector (G.993.5)

Do you have any idea about why they are not shown anymore?

gnuton, 2022-01-04:
I am not sure yet why these have been removed. Sometimes they are hidden in the page.
I should check the code.

BTW I have built a new image with updated ADSL driver
Please test this if you have some time and let me know if this fixes the problem.
https://1776-175516287-gh.circle-artifacts.com/0/workspace/release/DSL-AX82U_386.4_beta3_cferom_puresqubi.w
thanks al lot

JustifYI, 2022-01-04:
Just tested and YES, the SNR Adjustment is working now! :)

[SNR Adj. working]
[image]

Just one last question before closing: I've tried to build the FW by myself following your guide, but I got stuck on the "permissions" while trying to clone the git. Following the troubleshooting, the "id" cmd is not working on PowerShell. Do I need to use a Linux environment? 

(I apologize for the naive questions, but I'm new to this environment)

[git permissions]
[image]

gnuton, 2022-01-04:
the problem you have is with your github rights.
But I do not think you can sucesfully clone on the repo on a windows FS (which is not case sensitive) and compile it in a linux (case sensitive) environment. 
You will get into troubles.
To build asuswrt I suggest you VMWare or a linux machine.. you can test also WSL/2.
I will close this.
The new driver is in master so it will come withing the next images too.
thanks again.

closed 2022-01-04 by gnuton