gnuton/asuswrt-merlin.ng #139: VPN not working on Dual WAN Load Balancing mode

MHKalantarian, 2021-12-26:
**Firmware Version Affected **
386.03_2-gnuton5 

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**
When you enable VPN on Dual WAN Load Balancing mode, none of the connections will go through VPN specially when you have routing for splitting WANs, I have checked the ip rule table and the problem is when turning on VPN, the VPN rule will go to the lowest priority ignoring everything. If you insert the VPN rule on top of Dual WAN Routing rules which would be "pref 0-100" then it would work and also if you can separate load balancer routing rules with space between them you can use separate VPNs for each route.

**To Reproduce**
Steps to reproduce the behavior:
Turn on Dual WAN
Load Balancing Mode
SSH
"ip rule"

**Expected behavior**
VPN rule inserted on top or ability to insert VPN rule in between routing rules

gnuton, 2021-12-26:
Can you please also specifie the router model and the link to merlin issue?
thanks

MHKalantarian, 2021-12-26:
DSL-AC38U 
https://www.snbforums.com/threads/ovpn-when-dual-wan-load-balancing-is-active.76275/

gnuton, 2021-12-28:
This enanchment should come from upstream Merlin code.