gnuton/asuswrt-merlin.ng #140: New Models to support for 2022

gnuton, 2022-01-01:
Hi,
In 2022 I do have plans to support new models.
I do already have a few ones planned, but I can consider to take more if I do have enough interest in some specific models.

To propose your model, please search in the comments to check if your model is already listed and thumb it up.
If the model is not present yet:
1. Check the latest firmware version of your router. It has to be 386.xx.
2. Add a new comment to this thread. Please, do not open a new thread.

gnuton, 2022-01-01:
ZenWifi XT8

gnuton, 2022-01-01:
[ZenWifi ET8](https://www.asus.com/Networking-IoT-Servers/Whole-Home-Mesh-WiFi-System/ZenWiFi-WiFi-Systems/ASUS-ZenWiFi-ET8/)

Duckface21, 2022-01-02:
TUF-AX5400

Awakening1977, 2022-01-02:
TUF-AX5400

washran, 2022-01-02:
.@gnuton I'm using your asuswrt-merlin fork RT-AX82U. 386.4 beta 3 firmware.

icnmfabro, 2022-01-03:
ASUS RT-AC68U

gnuton, 2022-01-03:
@icnmfabro Asus RT-AC68U is already supported by the mainstream project: https://www.asuswrt-merlin.net/download

salavattvb, 2022-01-03:
Please make the firmware on TUF-AX5400

Rilly89, 2022-01-03:
RT-AX82U

gnuton, 2022-01-03:
> RT-AX82U

RT-AX82U is already supported this page is about adding new models beside the supported ones.

gnuton, 2022-01-04:
TUF-AX3000

Profile: TUF-AX3000 / HND-963178

jnsson, 2022-01-04:
RT-AX55 https://www.asus.com/se/Networking-IoT-Servers/WiFi-Routers/All-series/RT-AX55/ I have firmware 3.0.0.4.386_45375-ge5f218b on this router

Mariusz89B, 2022-01-04:
Asus RT-AC3200 this old router still have potential. :)

lamham25, 2022-01-06:
RT-AC86U
https://www.asus.com/Networking-IoT-Servers/WiFi-Routers/ASUS-Gaming-Routers/RT-AX86U/

acidleha78, 2022-01-06:
TUF-AX5400

jansenkarl, 2022-01-08:
How about the RT-AC67P? I just got one and was disappointed that the included firmware lacks the traffic analyzer section. It has the much less useful "traffic monitor." I upgraded from a very old AC68P and have been really depending on that feature.

markfi76, 2022-01-09:
RT-AX82U

Jacky10X, 2022-01-10:
RT-AX92U
https://www.asus.com/us/Networking-IoT-Servers/WiFi-Routers/ASUS-Gaming-Routers/RT-AX92U/

Jacky10X, 2022-01-11:
RT-AX55U:
https://www.asus.com/Networking-IoT-Servers/WiFi-Routers/All-series/RT-AX55/

Jacky10X, 2022-01-11:
RT-AX55:
https://www.asus.com/Networking-IoT-Servers/WiFi-Routers/All-series/RT-AX55/

BMiZEL, 2022-01-11:
RT-AX56U - 
I had to revert to v386_3_2 from v386_4_0, as the latest version has an unsolved bug, so a new bug-free alternative would be great

Duckface21, 2022-01-12:
A tip if u can, can u enable  theme change of GUI on tuf-5400 the new black theme sucks that asus  have. the classic theme is better

gnuton, 2022-01-12:
@Duckface21 I will check what I can do. I will tell you more once I do get the dev unit and the GPL for it.
GPL is coming next week.. so I may start to have some alpha release soon.. but I cannot test yet... so just for the brave ones...

gnuton, 2022-01-12:
@Duckface21 this is off-topic in this issue and  I will delete the message. 
I will create an issue for enable Merlin on the choosen routers.. there you can ask for requests.
In any case the blue theme will be the default one.

BK-Rider, 2022-01-16:
ROG STRIX GS-AX5400 - https://rog.asus.com/networking/rog-strix-gs-ax5400-model/

Current Version : 3.0.0.4.386_43588-g814828f

NathanBlais, 2022-01-17:
RX-AX89X -
https://www.asus.com/Networking-IoT-Servers/WiFi-Routers/ASUS-Gaming-Routers/RT-AX89X/

Current Version: 3.0.0.4.386.45130

Dziubek, 2022-01-28:
> RT-AX55 https://www.asus.com/se/Networking-IoT-Servers/WiFi-Routers/All-series/RT-AX55/ I have firmware 3.0.0.4.386_45934-gb23bdac on this router

If he receives support, this would be the most economical model AX :-D

gnuton, 2022-02-12:
> GT-AX6000
RMerlin has received one of them from Asus. So there may be chance he could support it.
https://twitter.com/RMerlinDev/status/1465762519490957324

JPatrick2398, 2022-03-08:
> TUF-AX3000
> 
> Profile: TUF-AX3000 / HND-963178

@yk00365 leave a like also above, there's a comment for TUF-AX3000 already

Argus04, 2022-03-08:
> TUF-AX3000
> 
> Profile: TUF-AX3000 / HND-963178

This would be a wise move to also support this router since you're already in the process of supporting TUF-AX5400, same hardware, same TUF product line.

yk00365, 2022-03-08:
> > TUF-AX3000
> > Profile: TUF-AX3000 / HND-963178
> 
> @yk00365 leave a like also above to vote, there's a comment for TUF-AX3000 already

Many thanks.

yk00365, 2022-03-08:
> > TUF-AX3000
> > Profile: TUF-AX3000 / HND-963178
> 
> This would be a wise move to also support this router since you're already in the process of supporting TUF-AX5400, same hardware, same TUF product line.

Yes, the tuf-ax3000 users are too much,and we are waiting for the Merlin.ng firmware for the TUF-AX3000. Yesterday(GMT+8:00), I am very surprised for his project is supported the TUF-ax5400. SO, I think, if he add the support for TUF-ax3000, is is a very great thing.
I would like waiting for the support for the TUF-AX3000.

RodrigoTomeES, 2022-03-18:
GT-AC5300
Firmware 3.0.0.4.386_42643-g16dc577

IndigoBodolf, 2022-03-20:
DSL-AC88U / AC3100
Firmware v1.10.08 build593 Aug. 16, 2021
https://www.asus.com/uk/Networking-IoT-Servers/Modem-Routers/All-series/DSL-AC88U/
Any chance this may be supported in the future? Or know of any other devs working on it?

karlgezer, 2022-03-22:
> DSL-AC88U / AC3100
> Firmware v1.10.08 build593 Aug. 16, 2021
> https://www.asus.com/uk/Networking-IoT-Servers/Modem-Routers/All-series/DSL-AC88U/
> Any chance this may be supported in the future? Or know of any other devs working on it?

Also looking for the DSL-AC88U - popular in mainland Europe for people wanting to connect on specific devices to US/UK servers via VPN
https://www.asus.com/uk/Networking-IoT-Servers/Modem-Routers/All-series/DSL-AC88U/

gnuton, 2022-03-24:
@IndigoBodolf @karlgezer as stated above I can support only routers running AsusWRT. DSL-AC88U has 1.x firmware so it ca

IndigoBodolf, 2022-03-24:
@gnuton On the specs page for the router it does say that it runs AsusWRT, although admittedly the version number is different, but I was curious about the feasibility. Sorry to be a bother! Hopefully at some point the router will get better support from ASUS or elsewhere, a shame if not, as the hardware itself is quite good.

MaryP12, 2022-03-25:
TUF-AX3000 please

Version 3.0.0.4.386

https://www.asus.com/Networking-IoT-Servers/WiFi-Routers/ASUS-Gaming-Routers/TUF-Gaming-AX3000/HelpDesk_BIOS/

bilboyfrodo, 2022-03-30:
AC87U
https://www.asus.com/uk/Networking-IoT-Servers/WiFi-Routers/ASUS-WiFi-Routers/RTAC87U/

The last Merlin is 384.13_10.

AX58Uv2, 2022-03-31:
RT-AX58U V2

gnuton, 2022-03-31:
@AX58Uv2 isn't it already supported by RMerlin?

gnuton, 2022-03-31:
@bilboyfrodo look at https://github.com/gnuton/asuswrt-merlin.ng/tree/DEV_rt_ac87u it should have refreshed version of merlin.
if you want you can maintain it .. :D

rpcarvalheira, 2022-03-31:
@gnuton since it's been a few months now, any news of ax92u? :(
Pretty please? 😭

ixxeL2097, 2022-04-05:
> ROG STRIX GS-AX5400 - https://rog.asus.com/networking/rog-strix-gs-ax5400-model/
> 
> Current Version : 3.0.0.4.386_43588-g814828f

Definately need this one, would be awesome !!

ThaKevMan, 2022-04-09:
ROG STRIX GS-AX5400 

https://rog.asus.com/networking/rog-strix-gs-ax5400-model/

tony880321, 2022-04-09:
ASUS ZenWiFi AC (CT8)
Latest firmware： 3.0.0.4.386.46122 (2022/03/02)
https://www.asus.com/Networking-IoT-Servers/Whole-Home-Mesh-WiFi-System/ZenWiFi-WiFi-Systems/ZenWiFi-AC-CT8/
Is it possible to support？ thank you so much

jumpsmm7, 2022-04-10:
RT-AX89X Please?

gnuton, 2022-04-11:
@tony880321 Yes C8 could support Merlin.

elong618, 2022-04-13:
> Hi, In 2022 I do have plans to support new models. I already have a few ones planned, but I can consider to take more o swap some of them if the community has a lot of interests in some specific models.
> 
> So, Is your router in the list below?
> 
> * YES -> then thumb it up  👍
> * NO ->   is your router alrady supported by me (DSL/RT-AX82U or DSL-AC68U) or RMerlin (https://www.asuswrt-merlin.net/download) ?
>   
>   * YES -> then all  good you are lucky! :D nothing to do for you!
>   * NO -> is your router running firmware 386.x?
>     
>     * YES - please add a **new comment** below with router model and link to asus page
>     * NO - Merlin won't easily work on this router and i do not have plan to support it for sure. Sorry!
>   
>   Note: Since I wanna just have a clean list of models I will be removing comments which do not contain any model or meaningful info for this thread. Thanks for the understading.

elong618, 2022-04-13:
hi，EA6300  V1  want to merlin  thanks

jumpsmm7, 2022-04-14:
> Hi, In 2022 I do have plans to support new models. I already have a few ones planned, but I can consider to take more o swap some of them if the community has a lot of interests in some specific models.
> 
> So, Is your router in the list below?
> 
>     * YES -> then thumb it up  +1
> 
>     * NO ->   is your router alrady supported by me (DSL/RT-AX82U or DSL-AC68U) or RMerlin (https://www.asuswrt-merlin.net/download) ?
>       
>       * YES -> then all  good you are lucky! :D nothing to do for you!
>       * NO -> is your router running firmware 386.x?
>         
>         * YES - please add a **new comment** below with router model and link to asus page
>         * NO - Merlin won't easily work on this router and i do not have plan to support it for sure. Sorry!
>       
>       Note: Since I wanna just have a clean list of models I will be removing comments which do not contain any model or meaningful info for this thread. Thanks for the understading.

@gnuton 

How about the TUF-AX3000

EggMass, 2022-04-14:
Please support the Asus TUF Gaming AX3000 Router.
Thanks

foxamei, 2022-04-14:
How about the TUF-AX3000 ?

MryangyangLi, 2022-04-16:
Please support the Asus ZenWiFi XD6 Router.
Thank you very much ! 

https://www.asus.com/Networking-IoT-Servers/Whole-Home-Mesh-WiFi-System/ZenWiFi-WiFi-Systems/ASUS-ZenWiFi-XD6/

zhuanshicong, 2022-04-16:
R8500?

Jdreioe, 2022-04-16:
AX Mini (CD6) would be nice as an entry-point mesh system.

fan881217, 2022-04-16:
TUF-AX3000
Please.

LordPhoenix-ZA, 2022-04-16:
ASUS ZenWifi AX XD4 Mini

Latest firmwareas of today

Version 3.0.0.4.386.46061

[https://www.asus.com/za/Networking-IoT-Servers/Whole-Home-Mesh-WiFi-System/ZenWiFi-WiFi-Systems/ASUS-ZenWiFi-AX-Mini-XD4/HelpDesk_BIOS/](https://www.asus.com/za/Networking-IoT-Servers/Whole-Home-Mesh-WiFi-System/ZenWiFi-WiFi-Systems/ASUS-ZenWiFi-AX-Mini-XD4/HelpDesk_BIOS/)

RichMD, 2022-04-16:
> > ROG STRIX GS-AX5400 - https://rog.asus.com/networking/rog-strix-gs-ax5400-model/
> > Current Version : 3.0.0.4.386_43588-g814828f
> 
> Definately need this one, would be awesome !!
Current firmware 3.0.0.4.386.46061

tzgroute, 2022-04-17:
请支持华硕ZenWiFi XD6路由器。

最新产品：3.0.0.4.386.46859

https://www.asus.com/Networking-IoT-Servers/Whole-Home-Mesh-WiFi-System/ZenWiFi-WiFi-Systems/ASUS-ZenWiFi-XD6/

可以支持吗？非常感谢！

[184 more replies]

closed 2024-01-12 by gnuton