gnuton/asuswrt-merlin.ng #141: Cannot Add AiMesh Nodes with

prometheus247, 2022-01-02:
**Router Model Affected**
Models: DSL-AX82U with 2x RP-AX56

**Firmware Version Affected**
386.04_0-gnuton1_beta3

**Is this bug present in upstream Merlin releases too?**
Dont Know

**Describe the bug**
Cannot add or find any AiMesh Nodes via Main Home Screen

**To Reproduce**
1. Reset RP-AX56 AiMesh Node
2. Plug it in the direct vicinity of the router DSL-AX82U
3. Check to see the node being in paring mode (solid blue LED)
2. Search for it via Router Home Screen

**Expected behavior**
The RP-AX56 Node should list up in the result page

**Screenshots**
[image]
[image]

**Comment**
The ASUS pages indicate, that the firmware of each AiMesh Device needs to be on the "same" branch / level. So I am wondering, if and how the AiMesh is compatible with Merlin in general and your gnuton fork particularly?

washran, 2022-01-02:
My AiMesh router RT-AX82U pairs well with my AiMesh node RT-AC86U. I had to disable wifi6 and wpa3 to make it work. Both wireless and ethernet backhaul doing the job. I'm using gnuton's 386.04_0-gnuton1_beta3 firmware.

washran, 2022-01-02:
Is the RP-AX56 a repeater? Does it have AiMesh capability? Sorry for so many edits above. Something happened when I posted and went into the wind.

prometheus247, 2022-01-03:
> Is the RP-AX56 a repeater? 
Yes, well its an AiMesh Node Repeater and it works flawless with the OEM Firmware.

> Does it have AiMesh capability?
Yes.
> I assume you read this manual?
I red the manuel. 

I think my issue is, that the Repeater cannot be brought to Merlin, so my AiMesh doesnt work. Had to switch back to full non Merlin / OEM again. Otherwise, my Network is broken.

gnuton, 2022-01-04:
Hi,
AiMEsh on stock firmware is the very same as the one in Merlin.
If it doesn't work there is a bug or something is not right with the setup.

gnuton, 2022-01-05:
@prometheus247 I am checking this. Got the same with a dirty flashed DSL-AX82U...

gnuton, 2022-01-05:
@washran have you set up your RT-AX82U as master node or slave?

washran, 2022-01-05:
> @washran
> 
>     * have you set up your RT-AX82U as master node or slave?
> 
>     * have you built your image by gnuton-master or another branch?

RT-AX82U master with gnuton's 386.4_b3
RT-AC86U slave with merlin's 386.4 release
Works excellent.

gnuton, 2022-01-05:
@washran and you are linking them via wireless and not over ethernet, right? 
Thanks I will double check now if something went wrong when i cleaned up the repository..

gnuton, 2022-01-05:
So I managed to link via AiMesh:
 - RT-AX82U with 386.4 beta3 as master
 - ZenWifi ET8 running stock firmware as node
The RT-AX82U was not able to find the ET8 node via wireless, but it was able via ethernet cable.
Ethernet cable may be  needed for the first setup. BTW confirmed that AiMesh works on RT-AX82U

[image]

gnuton, 2022-01-05:
I have tested also the DSL-AX82U and I had same experience than in RT-AX82U.
- DSL-AX82U was not able to find the node via wireless
- DSL-AX82U was able to find the node and connect to the node via ethernet.
Since the cable is needed only for the first setup I think using the cable is very fine.
After that the wireles uplink works great.

[image]

washran, 2022-01-05:
> So I managed to link via AiMesh:
>
>     * RT-AX82U with 386.4 beta3 as master
>
>     * ZenWifi ET8 running stock firmware as node
>       The RT-AX82U was not able to find the ET8 node via wireless, but [...]

I'm able to find the RT-AC86U easily via ethernet backhaul. But cannot set it up wirelessly. I can switch to wireless after it installs mesh node via ethernet. Then I can switch between wireless and ethernet as well as auto. I had to turn off wifi6 and wpa3 since the 86 router is kinda elderly. With stock firmware, I can start aimesh node wirelessly. With merlin and your firmware, ethernet setup aimesh works well. If both routers are ax models, you can use wifi6 and wpa3. Not really an issue... Just remember that AC and AX are oranges and apples, ha.

prometheus247, 2022-01-06:
Thats good news, will check to see if the ethernet backhaul setup works with mine too and come back to you.

lamham25, 2022-01-06:
Hello, it looks like I am having a similar problem. 
I have posted on snbforum as well ([here](https://www.snbforums.com/threads/setting-up-aimesh-system.76603/)) of my query:

Hello,

I have a DSL-AC68U router at home running on the merlin firmware (386.03_2-gnuton5).

I've been planning to set up an aimesh system at home to better my internet coverage, so I did some research and decided to purchase the RT-AC86U as the aimesh node.
Since I have a DSL-AC68U which has a modem integrated, I'm assuming that is the reason why I cannot use the DSL-AC68U as a node and it must be the main router.

I connected the RT-AC86U to my laptop to upload the latest merlin firmware (RT-AC86U_386.3_2_cferom_ubi.w) and in the administration tab, I selected it to be an aimesh node, which forwarded me to the automatic setup where it reboots the RT-AC86U and set it up as an aimesh.

Once rebooted, I reconnected my laptop to the AC68U to add the RT-AC86U as a aimesh node. It runs and loads up to 100%, but after, the end results indicate the following:
[image]

1. No one else is trying to set up an AiMesh system and I have waited 1 minute and tried again, but was still getting the same result.
2. The node router is next to the main router.
3. node router is powered on.
4. Aimesh node has been upgraded to RT-AC86U_386.3_2.
5. AiMesh node should have automatically reset to default through the automatic step.

Have I setup the aimesh system incorrectly?

I did contact asus support, and they were very friendly. We went through sets of rebooting the RT-AC86U router, but it still didn't work. He suspects that it may be the merlin firmware?

gnuton, 2022-01-06:
@lamham25 I replied to you. Solution use the ethernet cable for the setup.
I tested the beta3 last night and it worked.
here some screenshots to prove it https://github.com/gnuton/asuswrt-merlin.ng/issues/142

lamham25, 2022-01-07:
@gnuton thank you. I saw your reply on snbforum and gave it a try. First time it failed, but second time it worked. As you mentioned in #142 , yes, I had that issue when the AiMesh node was loading super fast first time installing.

My setup: I have left the RT-AC86U as asus stock firmware and my DSL-AC68U as merlin firmware. They seem to work well together so far.

Thank you for your reply. Sorry for double spamming. When it comes to technology, I am impatient.

gnuton, 2022-01-07:
@lamham25 which version have you used? stable or beta?

prometheus247, 2022-01-07:
That worked brilliant! Dirty Flash back to Merlin btw. kept the so far connected Ai Nodes too, so this would be another really one time alike workaround too.

But ofc, I did another reset of the Node and voilá:

[image]

[image]

From my POV this issue is resolved.

gnuton, 2022-01-07:
cool I will close it.

closed 2022-01-07 by gnuton

lamham25, 2022-01-07:
@gnuton 
I am using the stable merlin firmware. I have also upgraded my node firmware to a stable merlin firmware and it works fine.