gnuton/asuswrt-merlin.ng #144: Firmware upgrade is only accessible through Optus server.

gnuton, 2022-01-06:
Check why this code is still here. 

if(isSupport("is_ax5400_i1"))
{
$("#fw_note2").hide();
$("#fw_note3").show();
$("#fw_note3").html("Firmware upgrade is only accessible through Optus server."); //Untranslated
}
my file
https://github.com/gnuton/asuswrt-merlin.ng/commits/gnuton-master/release/src/router/www/Advanced_FirmwareUpgrade_Content.asp

upstream
https://github.com/RMerl/asuswrt-merlin.ng/commits/master/release/src/router/www/Advanced_FirmwareUpgrade_Content.asp

gnuton, 2022-01-07:
280f469c8d438104c4519d2542e7c57ebe1995d2 fixes it

DJE007, 2022-01-10:
Yes I confirm it's fixed in 386.04_0-gnuton1_beta4, thank you

gnuton, 2022-01-10:
awesome! I will close this.
Thakns a lot for your support!

closed 2022-01-10 by gnuton