gnuton/asuswrt-merlin.ng #145: AiMesh pairing works only with ethernet backhaul

gnuton, 2022-01-07:
Firmware 386.4

Issue:
- AX routers: node scanning cannot detect any node
- AC rouers: node scanning works, but pairing fails. Progression bar goes fast.

Currently AiMesh setup works only if the routers are connected together via cable during the setup.
Cable should be connected like this:
   Master LAN port -----> Node WAN port

gnuton, 2022-01-21:
Testing this on RT-AX82U now.
Web ui returns this:
{
"get_onboardinglist":{}
}

{
"get_onboardingstatus":{"cfg_obstatus":"2","cfg_obresult":"","cfg_newre":"","cfg_wifi_quality":"-65","cfg_obstart":"","cfg_obcurrent":"1642802109","cfg_obtimeout":"","cfg_obmodel":"","cfg_ui_obmodel":"","cfg_obrssi":"-16","cfg_obcount":"0","cfg_obreboottime":"","cfg_obstage":"0","cfg_obfailresult":"0","cfg_re_maxnum":"9","cfg_recount":"0","cfg_ready":"1"}
}

gnuton, 2022-01-21:
This is the logs for a working onboarding

{
"get_onboardinglist":{"FC:34:97:05:F5:E0":{"FC:34:97:3B:34:A0":{"model_name":"ZenWiFi_ET8","ui_model_name":"ZenWiFi ET8","rssi":-19,"source":1,"tcode":"US/01","type":0,"misc_info":{}}}}
}

{
"get_onboardingstatus":{"cfg_obstatus":"2","cfg_obresult":"","cfg_newre":"","cfg_wifi_quality":"-65","cfg_obstart":"","cfg_obcurrent":"1642803255","cfg_obtimeout":"","cfg_obmodel":"","cfg_ui_obmodel":"","cfg_obrssi":"-16","cfg_obcount":"0","cfg_obreboottime":"","cfg_obstage":"0","cfg_obfailresult":"0","cfg_re_maxnum":"9","cfg_recount":"0","cfg_ready":"1"}
}

gnuton, 2022-01-21:
let's analyze why get_onboardinglist doesn't return anything:
reason 1: this data gets populated by a json file ONBOARDING_LIST_JSON_PATH

gnuton, 2022-01-21:
on a working router cfg_server prints out this when aimesh node scan finds a new node:

gnuton@RT-AX82U-F5E0:/tmp/home/root# amas-utils[amas-utils-int.h(LINE:1996)]:TRACE LINE!!!
amas-utils[amas-utils-int.h(LINE:1997)]:LLDP_CUSTOM_TLV_CFG_SET:1997  lldpcli configure lldp custom-tlv replace oui F8,32,E4 subtype 7 oui-info FF,FF,FF,FF,FF,FF

amas-utils[amas-utils-int.h(LINE:1999)]:TRACE LINE!!!

gnuton, 2022-01-22:
$ rgrep lldpcli  *
lan.c:				eval("lldpcli", "configure", "system", "interface", "pattern", bind_ifnames);
rc.h:extern int lldpcli_det(void);
rc.h:extern int lldpcli_det2(void);
services.c:	eval("lldpcli","configure","system","interface","pattern",buf);
services.c:	eval("lldpcli","configure","lldp","tx-interval","10");
services.c:	eval("lldpcli","configure","lldp","tx-hold","2");
services.c:		sprintf(exec_lldpd, "lldpd -L /usr/sbin/lldpcli -I %s -s %s -dddd &", bind_ifnames, productid);
services.c:		fprintf(fp, "lldpd -L /usr/sbin/lldpcli -I %s -s %s\n", bind_ifnames, productid);
services.c:	fprintf(fp, "lldpcli configure lldp tx-interval 10\n");
services.c:	fprintf(fp, "lldpcli configure lldp tx-hold 2\n");
services.c:		fprintf(fp, "lldpcli configure system hostname %s\n", productid);
services.c:		fprintf(fp, "lldpcli configure system hostname Ai-Mesh\n");
services.c:	fprintf(fp, "lldpcli resume\n");
services.c:	killall_tk("lldpcli");

sugar012, 2023-09-15:
> Firmware 386.4
>
> Issue:
>
> * AX routers: node scanning cannot detect any node
> * AC rouers: node scanning works, but pairing fails. Progression bar goes fast.
>
> Currently AiMesh setup works only if the [...]

Hello! I'm trying to use my device to be a node in an aimesh with a TUF-AX3000. As you can imagine the TUF will be the master and the DSL doesen't have  a WAN port. right? Does things changed in the meantime ?

gnuton, 2023-09-15:
No the dsl-ac68u firmware doesn't have a wan to be used as node yet.