gnuton/asuswrt-merlin.ng #146: RT-AX82U with QoS set to 'cake' halves my download speed (from 500mbps to ~250mbps)

dcominottim, 2022-01-10:
Using the latest beta 386.04_0-gnuton1_beta4. I have a 500mbps download link that gets the full speed with the FW when using Adaptive QoS, but after setting 'cake' I can't get past 250mbps downloads (Download Bandwidth manually set to 500.00 Mb/s).

dcominottim, 2022-01-10:
Also, judging from the https://www.waveform.com/tools/bufferbloat test, it's as if cake isn't actually applied at all -- the download and upload latency don't improve vs Adaptive QoS, differently from when using cake on my Mikrotik router.

gnuton, 2022-01-10:
Hi and thanks for reporting this.
If you SSH onto the router and your run the following command do you see something like this?

# tc qdisc show | grep cake
qdisc cake 8001: dev br0 root refcnt 2 bandwidth unlimited diffserv3 triple-isolate nonat nowash no-ack-filter split-gso rtt 100ms raw overhead 0

dcominottim, 2022-01-10:
> Hi and thanks for reporting this. If you SSH onto the router and your run the following command do you see something like this?
>
> ```
> # tc qdisc show | grep cake
> qdisc cake 8001: dev br0 root refcnt [...]

Thanks for reaching out!

Below is the output I get (I have just 1x WAN and 1x PC eth cables connected):

```
qdisc cake 8001: dev eth4 root refcnt 2 bandwidth 46080Kbit diffserv3 dual-srchost nat nowash no-ack-filter split-gso rtt 100ms noatm overhead 18 mpu 64
qdisc cake 8002: dev ifb4eth4 root refcnt 2 bandwidth 512Mbit besteffort dual-dsthost nat wash ingress no-ack-filter split-gso rtt 100ms noatm overhead 18 mpu 64
```

gnuton, 2022-01-10:
cool so the kernel module is built and the rules gets applied to eth4 (8001 for up and 8002 for download)
I am not expert of cake, but if eth4 is the WAN these two rules  may be right.

I think the real issue that you are facing is the fact that cake is not for you... infact it's not recommended for connection up/down of 250Mbps or higher.

closed 2022-01-13 by gnuton

dcominottim, 2022-01-13:
> I think the real issue that you are facing is the fact that cake is not for you... infact it's not recommended for connection up/down of 250Mbps or higher.

Thanks for the reply.

I am having some trouble when trying to grasp the part I quoted from your comment. The issues that CAKE solves cannot be solved by faster links -- as can be read in http://www.bufferbloat.net (such as in "but link speed isn’t the problem - it’s your router buffering more data than necessary") and the authors' comments around the web. Therefore, I can only think you might have meant that 1) Asuswrt-Merlin/your fork cannot deal with >250Mpbs links and/or 2) RT-AX82U's hardware is not enough for CAKE + >250Mpbs links. Is that correct?

My bufferfloat tests and gaming experience are dramatically improved when using CAKE with my Mikrotik hAP ac3 router (and this also applies versus the AX82U with or without QoS). 

I have a hard time believing the RX-AX82U's CPU (3x Cortex A7 at 1.5 GHz) cannot do CAKE at my speeds but the ac3's CPU (4x Cortex A7 at 716 MHz) can.

gnuton, 2022-01-13:
Hi,
I am not using cake and I am not expert on it.

If you wanna get more concrete answers on why cake is not performing well on merlin you can try to reach ttgapers.
Here is his github page:
https://github.com/ttgapers/cakeqos-merlin

In the first page he states about this permance issue. you can ask him why this happens.
Feel free to let me know if there is anything I can do for fixing it.. even if those changes usually come from Upstream Merlin.