gnuton/asuswrt-merlin.ng #147: Support for TUF-AX5400

gnuton, 2022-01-12:
This thread is about supporting TUF-AX5400.
Feel free to ask questions here. 

MVP status:
- [ ] received GPL
- [ ] received development unit
- [ ] Merged GPL to feature-branch
- [ ] Making

ETAs:
alpha: TBD
beta:TBD
stable: TBD

salavattvb, 2022-01-12:
We are very much waiting for a new firmware from you.

Duckface21, 2022-01-12:
Is it possible to enable theme shange  so we can switch to  orginal Gray theme insted of the new black one its hard to se on the new asus black theme

gnuton, 2022-01-13:
The first version is going to be just plain merln. so most probably the theme will be the standard blue one.
@salavattvb can you elaborate more about WiFi gain restrictions?

salavattvb, 2022-01-13:
> can you elaborate more about WiFi gain restrictions?
For Europe, wi-fi power has been cut. For China, wi-fi is not cut and coverage is greater. I apologize for the Google translation)

DPandMe, 2022-01-13:
SMB 2.1, 3.0 Plz !!!

gnuton, 2022-01-13:
@DPandMe SMB upgrade will benefit all routers, for this reason this should come from upstram Merlin. 
I am more keen to add model specific changes.
@salavattvb AFAIK this is not something I can unlock since the firmwareis closed source

Duckface21, 2022-01-13:
an estimate when do you tink the  merlin firmware is out in first beta os so?

gnuton, 2022-01-13:
@Duckface21 It depends on when I get the device.. I hope to have the beta by end of Feb.

gnuton, 2022-01-25:
Update: 
- GPL received.
- The dev unit is scheduled to arrive during middle of Feb.

gnuton, 2022-02-07:
Update:

- The GPL has been partially merged and it builds and it's in GIT. WWW files must be cherry-picked but I do not have a development unit yet for testing, so I will do that later.

gnuton, 2022-02-16:
Update:

- I got the dev unit from Asus

- I have merged the GPL into Merlin base code and I do have[ a working branch for TUF-AX5400](https://github.com/gnuton/asuswrt-merlin.ng/tree/DEV_TUF_AX5400) now. I will be releasing the alpha once I do fix most of biggest issues

gnuton, 2022-02-18:
**PRE-ALPHA TESTING PHASE**

## versions:
1. First[ pre-alpha image is available](https://1930-175516287-gh.circle-artifacts.com/0/workspace/release/TUF-AX5400_386.4_0_cferom_puresqubi.w). PLease do not look at versioning in the page yet.

## WHAT DOESN'T WORK
- [ ] AURA RGB is not working. not able to choose RGB color due to missing ledg.o in the GPL
- [ ] speedtest doesn't work
- [ ] Missing VPN Director

## Basic things to be tested
- [ ] various QOS
- [ ] AIMesh
- [ ] TrendMicro
- [ ] VPNs
- [ ] amtm
- [ ] guest networks
- [ ]

gnuton, 2022-02-18:
@salavattvb wireguard will come from upstream merlin that is already planned.

salavattvb, 2022-02-18:
[image]
Спасибо. Я установил и тестирую. Ззаметил  что не показывает полоски.

salavattvb, 2022-02-19:
Feb 19 09:17:04 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #4094 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
Feb 19 09:17:04 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #4095 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
Feb 19 09:17:04 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #4096 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
Feb 19 09:17:04 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #4097 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
Feb 19 09:17:04 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #4098 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
Feb 19 09:17:04 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #4099 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
Feb 19 09:17:04 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #4100 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
Feb 19 09:17:04 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #4101 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
Feb 19 09:17:04 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #4102 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
Feb 19 09:17:04 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #4103 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
Feb 19 09:17:04 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #4104 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
Feb 19 09:17:04 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #4105 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
Feb 19 09:17:04 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #4106 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
Feb 19 09:17:04 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #4107 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
Feb 19 09:17:04 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #4108 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
Feb 19 09:17:04 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #4109 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
Feb 19 09:17:04 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #4110 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
Feb 19 09:17:04 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #4111 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
Feb 19 09:17:04 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #4112 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
Feb 19 09:17:04 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #4113 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
Feb 19 09:17:04 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #4114 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
Feb 19 09:17:04 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #4115 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
Feb 19 09:17:04 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #4116 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
Feb 19 09:17:04 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #4117 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
[... cut ...]
Feb 19 09:18:18 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #9581 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
Feb 19 09:18:18 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #9582 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings
Feb 19 09:18:18 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #9583 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings

VyachesDram, 2022-02-19:
Make a brightness adjustment LEDs.

gnuton, 2022-02-19:
@VyachesDram LED is completely broken in this build. I am wairing for a fix from ASUS.

gnuton, 2022-02-21:
@salavattvb as for the 
'''Feb 19 09:18:18 ovpn-client1[4215]: AEAD Decrypt error: bad packet ID (may be a replay): [ #9583 ] -- see the man page entry for --no-replay and --replay-window for more info or silence this warning with --mute-replay-warnings'''
https://www.snbforums.com/threads/openvpn-client-expressvpn-on-asus-merlin-386-1-not-uploading-correctly-vpn-client-disconnects-after-midnight.70442/

vladut123, 2022-02-21:
> @VyachesDram LED is completely broken in this build. I am wairing for a fix from ASUS.

Waiting?

gnuton, 2022-02-21:
@vladut123 yes sorry for the typo. I am waiting for Asus to send me new GPLs.
They promomise I will get all new ones by the end of this week.

gnuton, 2022-03-03:
Beta available here https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/386.05_0-gnuton0_beta1

JPatrick2398, 2022-03-03:
Nice work @gnuton. Is there a chance that you'll also support TUF-AX3000? Or just the TUF AX-5400?
I have an AC68U and TUF-AX3000, I can say that AC68u with merlin firmware is much more stable than the TUF AX-3000 considering their hardware specifications, in which the TUF-AX3000 has a higher specs than the AC68u, firmware really has a greater impact.

gnuton, 2022-03-03:
Please add your thumb up here. There is chance for me to support more
models, but at moment no plans yet.
https://github.com/gnuton/asuswrt-merlin.ng/issues/140

gizmosngadgetz, 2022-03-10:
Thanks for your hardwork and contributions! I can't seem to find the DNSfilter tab under LAN. Is it hidden away somewhere else or not included in this beta? 

The router also doesn't seem to read my nat-start script in /jffs/scripts/ ...though perhaps this is my fault since its my first time doing this. I've made the file executable and used Notepad++ so it had proper UNIX headers. Anything else I should check?

gnuton, 2022-03-14:
d1e82842ee fixes DL/UP invisible bars. Fix in the next beta.

gnuton, 2022-03-14:
@gizmosngadgetz DNSFilter fixed (more info here https://github.com/gnuton/asuswrt-merlin.ng/issues/169)

pyadvichuk, 2022-03-20:
Nice work @gnuton, thanks a bunch. Is it possible to add `usbip` host modules to the build? like
```
CONFIG_USBIP_CORE=y
CONFIG_USBIP_HOST=y
```

Absent83, 2022-03-22:
Is there any information about the **OpenVPN client**? Does it work?

gnuton, 2022-03-23:
@pyadvichuk please open a ticket for that since it's an enanchement. 
Please describe the reason and why other people would need that too.

gnuton, 2022-03-23:
@Absent83 I have not tested that but TUF-AX5400 but if that's a merlin option you should have it on my images too and if it's not there please ping me

Absent83, 2022-03-23:
I have bought TUF-AX5400 and updated with **Unstable 386.05_0-gnuton0_beta1 Pre-release**

There is a parameter, that I mentioned before.

[image]

I will check how it works.

Fatmajk, 2022-03-24:
Amazing work @gnuton . Highly appreciated works, really. Looking forward to see a stable build! 
**Thanks a lot!**

swiffer, 2022-03-25:
Tested OpenVPN Client Setup after flashing the Firmware (works finde!) However VPN Director Policy Based Routing ist Missing From the GUI as Well as VPN Status

 https://github.com/RMerl/asuswrt-merlin.ng/wiki/VPN-Director

gizmosngadgetz, 2022-03-26:
> @gizmosngadgetz DNSFilter fixed (more info here #169)

Thanks! Reading through your changes log, I realized the feature is activated and I can access it directly through .../DNSFilter.asp

Likewise, I noticed that **Advanced VPN Director** tab is missing, but is an activated feature that I can access directly through .../Advanced_VPNDirector.asp

swiffer, 2022-03-26:
@gizmosngadgetz thanks! Works like a charm!

Bilk0, 2022-03-26:
Flashed it last night and am impressed how stable  it is for a beta release.  So far so good. Thank you for your hard work.

gnuton, 2022-03-26:
Stable released! Enjoy!

closed 2022-03-26 by gnuton

Bilk0, 2022-03-27:
Installed the stable release and its working well, no issues identified and thank you once again for your hard work in getting this working on the AX5400.

RichMD, 2022-03-28:
Will the GS-AX5400 be supported (almost identical hardware)?

gnuton, 2022-03-28:
@RichMD no plans as for now.
Please add your vote here https://github.com/gnuton/asuswrt-merlin.ng/issues/140 btw

Bilk0, 2022-03-29:
One thing I have noted is that RAM usage appears to be lower using this Merlin Firmware, on stock FW I would often see only around 55MB free, with this FW this has risen to around 150MB free.

Duckface21, 2022-04-01:
is it possible to add support fpr NTP server in merlin så  every comp sync time  whit router

gnuton, 2022-04-01:
@Duckface21 ilocal NTP server is already in the image. you can enable from http://router.asus.com/Advanced_System_Content.asp

fededim, 2022-04-02:
Flashed yesterday, custom hosts file works perfectly, @gnuton thank you very much for this release!

Bilk0, 2022-04-06:
@gnuton , 

I have VPN Client on, (using VPN Director to assign various devices to WAN or VPN), I have checked and my VPN is working. 
I also have AI Protection on , QOS (Adaptive) on, Traffic Analyser On, LAG on and a USB RAID connected to USB port, but RAM usage is still lower than stock FW (Currently have 130MB free).  Its been rock solid , I fact, I was just testing the AX5400 as I have a complete Unifi setup on my home network (USG Pro 4 (Upgraded to 4GB), 6 switches, 4x APs , Unfi Protect with 4 cameras), but I have deactivated the USG Pro and 2xAPs and am using the AX5400 as I am liking the additional features. 
I am very pleased with the firmware you have provided, kudos for doing such a great job and to Merlin of course.

fauxfongers, 2022-08-09:
> Stable released! Enjoy!

Is there a reason why ax5400 isn't on SourceForge, but is available on GitHub?
At first I saw only the Alpha cause it didn't appear on merlin website nor on SF.

gnuton, 2022-08-10:
@fauxfongers 
This github is maintained by a different person.
SF is the main project and he maintains some models, but he cannot maintain all of them.
I have started with DSL routers but then I got asked by the lead merlin dev to support more routers..

ajdee2, 2022-10-03:
i have TUF-AX5400 is't any working stable version to this router ?
and is there any version that has support for wireguard in vpn

Duckface21, 2023-11-12:
HI under Open VPN server is there a chanse to enable more then one Server instance 2 is not the best for my use

gnuton, 2023-11-21:
It was decreased in order to avoid stability issue due to low RAM

Duckface21, 2023-11-21:
oki but wold it be possible to enable more  but add a stability warning whit more then 2 running so usersn know

gnuton, 2023-11-21:
I am not sure I need to check it. This was an upstream change btw, so other merlin routers with latest firmware have same limitation.

Duckface21, 2023-11-21:
oki wold be nice if u can check