gnuton/asuswrt-merlin.ng #148: Feedback for 386.04_0-gnuton1_beta4 running on RT-AX82U

anarchy9388, 2022-01-12:
**Router Model Affected**
RT-AX82U

**Firmware Version Affected**
386.04_0-gnuton1_beta4

**Feedback**
Everything seems to be working well, I'm using PPPoE, QoS etc all the advanced features and haven't hit any issues yet.

Only thing i do see in the logs is this, but not sure if its really an issue or not:
```
Jan 12 12:00:48 kernel: nf_conntrack: expectation table full
Jan 12 12:00:48 kernel: nf_conntrack: expectation table full
Jan 12 12:00:48 kernel: nf_conntrack: expectation table full
Jan 12 12:00:48 kernel: nf_conntrack: expectation table full
Jan 12 12:00:48 kernel: nf_conntrack: expectation table full
Jan 12 12:00:51 kernel: nf_conntrack: expectation table full
Jan 12 12:00:53 kernel: nf_conntrack: expectation table full
Jan 12 12:00:53 kernel: nf_conntrack: expectation table full
Jan 12 12:00:55 kernel: nf_conntrack: expectation table full
Jan 12 12:00:55 kernel: nf_conntrack: expectation table full
Jan 12 12:00:55 kernel: nf_conntrack: expectation table full
Jan 12 12:00:55 kernel: nf_conntrack: expectation table full
Jan 12 12:00:55 kernel: nf_conntrack: expectation table full
Jan 12 12:00:55 kernel: nf_conntrack: expectation table full
Jan 12 12:00:56 kernel: nf_conntrack: expectation table full
Jan 12 12:00:56 kernel: nf_conntrack: expectation table full
Jan 12 12:01:28 kernel: nf_conntrack: expectation table full
Jan 12 12:01:28 kernel: nf_conntrack: expectation table full
Jan 12 12:01:37 kernel: nf_conntrack: expectation table full
Jan 12 12:01:37 kernel: nf_conntrack: expectation table full
```

@gnuton ill continue to provide more feedback/details as i run the firmware for longer. thanks!

gnuton, 2022-01-13:
This is a known issue:
workaround here:
https://www.snbforums.com/threads/nf_conntrack-expectation-table-full-and-other-log-oddities.55415/page-4

anarchy9388, 2022-01-19:
Thanks @gnuton so far so good, not having any issues yet. :)

anarchy9388, 2022-01-24:
Closing this for now, haven't found any issues.

anarchy9388, 2022-01-24:
Keep seeing this in the logs and my ssh connections getting dropped a lot

```
Jan 24 16:45:16 RT-AX82U-F428 kernel: [<c0022658>] (do_work_pending) from [<c001f4cc>] (work_pending+0xc/0x20)
Jan 24 16:45:34 RT-AX82U-F428 kernel: potentially unexpected fatal signal 11.
Jan 24 16:45:34 RT-AX82U-F428 kernel: CPU: 0 PID: 4018 Comm: diversion Tainted: P           O    4.1.52 #1
Jan 24 16:45:34 RT-AX82U-F428 kernel: Hardware name: Generic DT based system
Jan 24 16:45:34 RT-AX82U-F428 kernel: task: d3d35400 ti: cd956000 task.ti: cd956000
Jan 24 16:45:34 RT-AX82U-F428 kernel: PC is at 0x3919c
Jan 24 16:45:34 RT-AX82U-F428 kernel: LR is at 0x39374
Jan 24 16:45:34 RT-AX82U-F428 kernel: pc : [<0003919c>]    lr : [<00039374>]    psr: 80080010
Jan 24 16:45:34 RT-AX82U-F428 kernel: sp : be9bce58  ip : be9bd0a8  fp : 00cb683c
Jan 24 16:45:34 RT-AX82U-F428 kernel: r10: 00000001  r9 : 00c9824c  r8 : 00000000
Jan 24 16:45:34 RT-AX82U-F428 kernel: r7 : ffffffff  r6 : 00000000  r5 : 00c5a540  r4 : 00000002
Jan 24 16:45:34 RT-AX82U-F428 kernel: r3 : 00066e58  r2 : 001fb0d6  r1 : 00000000  r0 : 00c9832c
Jan 24 16:45:34 RT-AX82U-F428 kernel: Flags: Nzcv  IRQs on  FIQs on  Mode USER_32  ISA ARM  Segment user
Jan 24 16:45:34 RT-AX82U-F428 kernel: Control: 10c5387d  Table: 0a9b004a  DAC: 00000015
Jan 24 16:45:34 RT-AX82U-F428 kernel: CPU: 0 PID: 4018 Comm: diversion Tainted: P           O    4.1.52 #1
Jan 24 16:45:34 RT-AX82U-F428 kernel: Hardware name: Generic DT based system
Jan 24 16:45:34 RT-AX82U-F428 kernel: [<c0026e60>] (unwind_backtrace) from [<c0022c38>] (show_stack+0x10/0x14)
Jan 24 16:45:34 RT-AX82U-F428 kernel: [<c0022c38>] (show_stack) from [<c0450ad4>] (dump_stack+0x8c/0xa0)
Jan 24 16:45:34 RT-AX82U-F428 kernel: [<c0450ad4>] (dump_stack) from [<c003aacc>] (get_signal+0x490/0x558)
Jan 24 16:45:34 RT-AX82U-F428 kernel: [<c003aacc>] (get_signal) from [<c00221d0>] (do_signal+0xc8/0x3ac)
Jan 24 16:45:34 RT-AX82U-F428 kernel: [<c00221d0>] (do_signal) from [<c0022658>] (do_work_pending+0x94/0xa4)
Jan 24 16:45:34 RT-AX82U-F428 kernel: [<c0022658>] (do_work_pending) from [<c001f4cc>] (work_pending+0xc/0x20)
```
[messages.log](https://github.com/gnuton/asuswrt-merlin.ng/files/7929628/messages.log)

gnuton, 2022-01-24:
is Trend Micro enabled? if you disable it do you still get this?

anarchy9388, 2022-01-24:
How can I check the Trend Micro status?

anarchy9388, 2022-01-24:
nevermind, just withdrew from trend micro, will see if it stops

anarchy9388, 2022-01-25:
appears to be an issue with "Diversion", will check further and update here

gnuton, 2022-01-25:
http://router.asus.com/AiProtection_HomeProtection.asp
set ai protection to off

anarchy9388, 2022-01-25:
It's gone now, was an issue with https://diversion.ch/ 
upgraded diversion and dont see the message anymore

gnuton, 2022-01-25:
may i close this?

anarchy9388, 2022-01-25:
Yup go for it 👍  Thanks!

closed 2022-01-25 by anarchy9388