gnuton/asuswrt-merlin.ng #149: CI: Automatic Builds

yknx4, 2022-01-19:
I am really grateful for your project and I'm a developer myself (although c++/c is like magic to me haha)

If you are interested I could help you setup Github Actions for automatic builds in case you are interested.

gnuton, 2022-01-20:
I would be interested in that since I do have problem with CircleCI that limits my builds to 1 hour.
There is no C++/C knowledge required. I do have a docker image with all stuff you need already in there.
Can github host runs longer than 1 hour?

yknx4, 2022-01-21:
It seems [so](https://docs.github.com/en/actions/learn-github-actions/usage-limits-billing-and-administration#usage-limits)
```
Job execution time - Each job in a workflow can run for up to 6 hours of execution time. If a job reaches this limit, the job is terminated and fails to complete.

Workflow run time - Each workflow run is limited to 72 hours. If a workflow run reaches this limit, the workflow run is cancelled.
```

gnuton, 2022-01-22:
Cool! :D
The task can get started then.
The task is about migrating the cirrent circle ci configuration to github actions.
The documentation for the migration is here: https://docs.github.com/en/actions/migrating-to-github-actions/migrating-from-circleci-to-github-actions
As for now the circleci configuration looks like 
[image]
but the first step was an optimization.. you can simplify it by checking out the code in each build.
and it can be found here:
https://github.com/gnuton/asuswrt-merlin.ng/blob/gnuton-master/.circleci/config.yml
shallow clone orbs is not used.

gnuton, 2022-01-22:
I think in git hub actions we have to define a matrix. but this is for later.
I would spit this task in several steps.
The first one is getting a single model compiled
We can start from DSL-AC68U

You can fork this repo and add your github action and try to get it working on your side. when it works you can send me a PR if you like.

I think initially the GA config could be something simple like this.
```
on: push
jobs:
  build_dslac68u:
    runs-on: gnuton/asuswrt-merlin-toolchains-docker:latest-ubuntu-20_04
    steps:
      - uses: actions/checkout@v2
      - run: |
        export MODEL="dsl-ac68u"
        export PROJECT_DIR="/home/docker/project"
        export RELEASE_DIR="src-rt-6.x.4708"
        export IMAGE_PATH=${PROJECT_DIR}/release/${RELEASE_DIR}/image/
        export MERLINUPDATE=y
        cd ${PROJECT_DIR}/release/${RELEASE_DIR}
        make ${MODEL}
        ls ${IMAGE_PATH}
        
```

gnuton, 2022-01-22:
Created small job https://github.com/gnuton/asuswrt-merlin.ng/actions/runs/1732772929 but somehow it's stuck.
No idea why it deosn't get picked up by any worker. Maybe I missing something in my config?

gnuton, 2022-01-25:
@yknx4 when you have some time please check the config I wrote.. no idea why it doesn't work yet :D

yknx4, 2022-01-25:
I've actually been working on a separate branch on it too. 

[image]

I've managed to built and output an artifact, based on your CircleCI config for the rt-ax82u, but when I added more devices, the compiler starts failing complaining about a missing `libmpc.so.2` that I'm use is in the LD_PATH, i've been quite stuck on that.

[image]

I've done a loooot of tests on how to load the paths hahaha

yknx4, 2022-01-25:
This is the code I have so far
https://github.com/yknx4/asuswrt-merlin.ng/blob/github-actions/.github/workflows/build.yml

gnuton, 2022-01-25:
the build has to run inside the docker image I have created.
is it running inside?
I cannot see any reference to the docker image.

yknx4, 2022-01-26:
@gnuton I didnt use the docker image, I copied the commands from your dockerfile to the install-deps and install-toolchain actions

gnuton, 2022-01-26:
ok. the installation seems fast..  46s is not bad.
I do still prefer docker so that I have same environment on the cloud as well as on my local machine :D

as for libmpc have you already look at https://github.com/gnuton/Asuswrt-Merlin-Toolchains-Docker/commit/1fde7dc8b13dcb90940454b165e98e7960a55ced ?

gnuton, 2024-01-26:
I have finally moved my building system to github actions.
I got some parts from your work @yknx4 thanks a lot.

The main differences are that I am still using docker because I wanted to have same working environment that I used for development.

Github actions vs Circle CI
- Circle CI allows more parallel job to run -> faster
- for single jobs Circle CI large is as large as the offered free default runner in Github
- Docker support is much better on Circle CI
- Debugging is much better on CircleCI since it allows opening an SSH session
- WEB UI is better in Circle CI since it shows the job which won't run in the diagram flow
- Logs are better in the Circle CI web UI, it can handle better 30 MB long logs
- The 1-hour cap on Circle CI is what forced me to move away from it.

closed 2024-01-26 by gnuton