gnuton/asuswrt-merlin.ng #150: Traffic monitoring slows down lan

Revenger, 2022-01-24:
**Router Model Affected**
Models: (eg: DSL-AC68U)

**Firmware Version Affected**
386.03_2-gnuton5

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**
Running traffic monitoring slows lan upload down speedtest showed significant speedloss.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn in traffic monitoring
2. Speedtest.net
3. Upload went from 25Mbps to 4MBPS
4. Turn off monitoring upload back to normal.

**Expected behavior**
No speed loss

**Screenshots**
Traffic Manager Off
https://imgur.com/XAr1zsB

Traffic Manager On
https://imgur.com/aQkopw7

Revenger, 2022-01-24:
Changed Stability and Dynamic Line Adjustment to off and the Annax from B to A and upload is fine on traffic manager both on and off.
Will keep a eye on things.

Could be one of those settings causing it.

gnuton, 2022-01-24:
Hi can you please test the beta firmware too. It should be stable now...

gnuton, 2022-01-24:
The next release will be based on the current beta and since it's fine I think this problem doesn't exist for it.
right?
if so can we close this task or is there any reason to keep this open? (you can always re-open it)

Revenger, 2022-01-24:
Upload is fine now so yea can close this and will reopen of needed.

closed 2022-01-24 by Revenger