gnuton/asuswrt-merlin.ng #151: DNSSEC Settings missing under WAN options

Randi1115, 2022-01-25:
**Router Model Affected**
Models: DSL-AC68U

**Firmware Version Affected**
386.04_gunton01

**Is this bug present in upstream Merlin releases too?**
N/A

**Describe the bug**
After the latest firmware upgrade, DNSSEC settings are missing from the WAN options. Can you please check why this is the case? Thanks.

gnuton, 2022-01-26:
[just for me]
I need to reapply this patch
https://github.com/gnuton/asuswrt-merlin.ng/commit/3829caeb851ab57ba2e545119a7fa45ef980a2d2#diff-456a4177443b2c80afe01e949cfbbdce1b0485a1d88b0df11cfee6fb00ccea9a

gnuton, 2022-01-27:
Test images:
those are safe to install just current stable + fix for DNS SEC + dns over tls
https://1832-175516287-gh.circle-artifacts.com/0/workspace/release/DSL-AX82U_386.4_0_cferom_puresqubi.w
https://1830-175516287-gh.circle-artifacts.com/0/workspace/release/DSL-AC68U_386.4_0_DSL_1.0.5.3.trx
Any feedback  is welcome

Randi1115, 2022-01-30:
Hi Gunton,

DNSSEC and DoT is working. Thanks.

gnuton, 2022-01-31:
Awesome I will be releasing this as stable.. :D

gnuton, 2022-01-31:
stable released

closed 2022-01-31 by gnuton