gnuton/asuswrt-merlin.ng #152: Support for DNS over HTTPS (DOH)?

HamidZaeri, 2022-01-26:
Hi

Does this firmware support DOH for DSL-Ac68U?
If yes, where is the option for it or how can i config it?

If no, is there any plan to implement this must have feature?

gnuton, 2022-01-26:
hi @HamidZaeri it's supported I think there has been a bad merge..
I will enable it with next release.

gnuton, 2022-01-26:
[just for me]
To fix this I need to reapply this patch
https://github.com/gnuton/asuswrt-merlin.ng/commit/f9c224d6896b8171c7934113a29f16550247dcf5#diff-456a4177443b2c80afe01e949cfbbdce1b0485a1d88b0df11cfee6fb00ccea9a

gnuton, 2022-01-27:
Test images:
those are safe to install just current stable + fix for DNS SEC + dns over tls
https://1832-175516287-gh.circle-artifacts.com/0/workspace/release/DSL-AX82U_386.4_0_cferom_puresqubi.w
https://1830-175516287-gh.circle-artifacts.com/0/workspace/release/DSL-AC68U_386.4_0_DSL_1.0.5.3.trx
Any feedback  is welcome

TrilIiard, 2022-01-27:
Everything seems to work great so far

gnuton, 2022-01-27:
@TrilIiard have you tested DOH?

TrilIiard, 2022-01-27:
Working, but cant test secure sni because i just dont know how

Hello-olleH, 2022-01-28:
> Test images: those are safe to install just current stable + fix for DNS SEC + dns over tls https://1832-175516287-gh.circle-artifacts.com/0/workspace/release/DSL-AX82U_386.4_0_cferom_puresqubi.w [...]

Before Update: 
[image]

After Update: 
[image]

Test Result: 
https://cloudflare-dns.com/help/
[image]

Thank You!

Hello-olleH, 2022-01-28:
A little bug: 
It says to me, there is an Update :D 

[image]

gnuton, 2022-01-30:
@Hello-olleH that's not a but.. that's fine. It just try to make you move back to stable since this is an internal build which has random version. 
Thanks a lot for testing

gnuton, 2022-01-31:
changes released within the latest stable: https://github.com/gnuton/asuswrt-merlin.ng/issues/152

closed 2022-01-31 by gnuton

HamidZaeri, 2022-02-12:
> changes released within the latest stable: #152

This feature is not supported for ADSL ?

Hello-olleH, 2022-02-20:
Disabled DoT but Server List can't be deleted

Hello-olleH, 2022-02-20:
@gnuton