gnuton/asuswrt-merlin.ng #154: 5 GHz Guest network 1 not connecting

adestounis, 2022-02-03:
**Router Model Affected**
Models: DSL-AC68U

**Firmware Version Affected**
386.4

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**
After the latest release 5ghz gn1 stopped assigning ip when access intranet is disabled.

adestounis, 2022-02-03:
I've not tried any other Merlin firmware except yours and i can confirm that the previous one was ok.
In order to clarify as much as possible, this issue corresponds to the original AC68U firmware.
There was a time when AiMesh v2 has first "introduced", the 2.5GHz band was affected by this problem (not 5 GHz) but in the latest ones, the problem was resolved.

fyi
https://www.snbforums.com/threads/asus-aimesh-guest-network-issues.72955/

gnuton, 2022-03-06:
can you test if the new beta has fixed this issue?

adestounis, 2022-03-08:
I'll try it and will be back with any feedback

adestounis, 2022-03-28:
Dear Antonio, it still does not assign IP when trying to connect. In the case where intranet access is enabled then it performs as it should.

gnuton, 2022-03-28:
Not the first time we issues with guest network and fixing those is not a piece of cake...
In the past I debugged the issue and I found out this was the problem https://github.com/gnuton/asuswrt-merlin.ng/commit/6700436db5fe34b056637e8c0ecbe1ee57ec2349
I am now building a new image without my fix just to see if this is still needed with the latest Asus pre-builds

gnuton, 2022-03-28:
if you like you can test  this image which is the stable without the above fix.
https://dl.circleci.com/private/output/job/f8c781db-207c-465e-91f2-c4c30b618d0b/artifacts/0/workspace/release/DSL-AC68U_386.5_2_DSL_1.0.5.3.trx

adestounis, 2022-03-30:
Thank you. By removing the fix from the changeList [6700436](https://github.com/gnuton/asuswrt-merlin.ng/commit/6700436db5fe34b056637e8c0ecbe1ee57ec2349) the 1st 5ghz guest network works again!!

adestounis, 2022-06-24:
Can you please revert the above change also in the future release 386.7?