gnuton/asuswrt-merlin.ng #156: 2.4ghz WiFi is not showing rt-ax82u

shrek1802, 2022-02-06:
**Router Model Affected**
Models: RT-AX82U

**Firmware Version Affected**
386.04_0-gnuton2

**Describe the bug**
So when rebooting the router there is no WiFi at all if I disable the smart connect then re enable I get WiFi. I have a few devices that can only use 2.4ghz and no matter what I do these cannot see the WiFi.

Also if I separate the band's to 2.4 and 5.0 the 2.4 and rename one the 2.4 is not showing up on my phone when scanning for WiFi only the 5.0.

gnuton, 2022-02-06:
too weird.. do you still get this issue if you reset to factory settings?

shrek1802, 2022-02-06:
Yes first thing I did was reset to factory settings

shrek1802, 2022-02-07:
Anyone think of anything I can try ?

gnuton, 2022-02-08:
1. does the stock firmware have the same issue? or is this something you see only with Merlin?
2. can you see the WiFi from a new device? or the wifi is not visible to any of your clients?
3. have you tried to change any parameter in the http://router.asus.com/Advanced_WAdvanced_Content.asp ? some old wireless adapter may have issues with power management

gnuton, 2022-02-08:
BTW I have bee told that some users are experiencing WiFi issues with IOT devices iwth the current GPL release.
46065 GPL may fix this.
We are currently receiing this new GPL from ASsus. Eric has already started the merge.

shrek1802, 2022-02-09:
Yeah funny u say that it is IOT devices I'm having the problem with with they all have tasmota firmware on them. I will wait for the update and let u no.  I just did understand after rebooting why there was no WiFi.

gnuton, 2022-02-09:
the new firmware will hopefully be in beta or alpha by the end of Feb or
middle of next month.
In the meantime you may double check if the stock firmware works for you..
If you do so that will help me a lot to understand if the problem is
because of Asus or because I did something wrong elsewhere.. but as for now
I do not have way for testing and checking this.

shrek1802, 2022-02-09:
I will flash the stock firmware and report back mate

vladut123, 2022-02-09:
> BTW I have bee told that some users are experiencing WiFi issues with IOT devices iwth the current GPL release.
> 46065 GPL may fix this.
> We are currently receiing this new GPL from ASsus. Eric has already started the merge for some devices.

iwth or with?

shrek1802, 2022-02-09:
So flashed the lastest stock firmware over the custom one you make and both channels when separated can be see from my phone and with all my IOT devices.

But with a little more digging what I have found of u flick the switch enable smart connect back on you can't see any WiFi. If you then goto quick internet setup and tell that to run the networks joined all comes back on and you can see wifi with everything again. That all with stock firmware

gnuton, 2022-02-10:
@shrek1802 thanks for checking this. You tested the latest stock firmware which is 46061.
The new stable will be most likely be based on 46065. which is pretty much the same as 61... but I do not have the full changelog yet.

shrek1802, 2022-02-12:
Not a problem fingers crossed the new one fixes the problems I'm having

shrek1802, 2022-03-02:
@gnuton sent u a little something on PayPal mate keep up the good work and any ETA on the new version that I'm hoping fixes this error I was getting ?

shrek1802, 2022-03-02:
Nice mate thank you

gnuton, 2022-03-26:
@shrek1802 can i close issue? The new stable is out and the issues with IoT devices may have been resolved by the newest drivers

closed 2022-03-28 by gnuton