gnuton/asuswrt-merlin.ng #158: Support for ZenWifi XT8

gnuton, 2022-02-21:
This thread is about supporting TUF-AX5400.
Feel free to ask questions here. 

Status:
- [x] received GPL
- [x] received development unit
- [ ] Merged GPL to feature-branch
- [ ] Initial testing & Bugfixing phase
- [ ] Public pre-alphas
- [ ] Alpha released
- [ ] collecting testing reports  & Bugfixing phase
- [ ] Beta released
- [ ] collecting testing reports  & Bugfixing phase
- [ ] Stable released

ETAs:
alpha: TBD
beta: TBD
stable: TBD

gnuton, 2022-03-02:
Note:
 The following sw has been added and prebuilds copy tool must be updated:
 - router/i2c-tools-2013-12-15
 - btconfig
 - bluez

gnuton, 2022-03-03:
Beta available here https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/386.05_0-gnuton0_beta1

sadoneli, 2022-03-06:
RT-AX95Q (ZenWiFi_AX6600) Tools_Sysinfo.asp page error

1. missing model RT-AX95Q: https://github.com/gnuton/asuswrt-merlin.ng/blob/gnuton-master/release/src/router/httpd/sysinfo.c#L632
2. missing model `MODEL_RTAX95Q`: https://github.com/gnuton/asuswrt-merlin.ng/blob/gnuton-master/release/src/router/httpd/sysinfo.c#L199

[image]

woudt, 2022-03-08:
One minor issue found, otherwise the firmware seems to werk perfectly fine for me!

The WiFi Radar option is shown on the menu, but leads to a page not found error.

[image]

[image]

gnuton, 2022-03-09:
I will try to fix this with the next beta

gnuton, 2022-03-14:
just for myself... the fix for this is similar as in a561e6a7a2a

gnuton, 2022-03-26:
all issues fixed and stable released!

closed 2022-03-26 by gnuton

axeloz, 2022-03-30:
Thanks a lot @gnuton for your work. I was waiting for this for quite some time. 
I'm just a little bit confused: you are talking of the XT8 but you wrote:
`This thread is about supporting TUF-AX5400.`

For me, the XT8 is AX6600 as confirmed here: https://www.asus.com/fr/Networking-IoT-Servers/Whole-Home-Mesh-WiFi-System/ZenWiFi-WiFi-Systems/ASUS-ZenWiFi-AX-XT8/techspec/
The one I own is the XT8 (B-1-PK) has shown on the link above. Is it compatible?

Thanks a lot

clementchong, 2022-04-09:
Possible to enable link aggregation or is this limited by hardware?

gnuton, 2022-05-17:
> This is extremely confusing. I am trying to run merlin on my XT8 and
> cannot seem to find any download link on the official website. Is there or
> is there no merlin version for the XT8?

XT8 merlin images are provided by me and I do release them only on github
and not on the website.
This is in any case an official fork.

sirdragonlord, 2022-08-26:
Hi,

Just got the XT8 twin pack and wondered if this fixes issues with guest networking, as asus firmware says supports all nodes, but doesn't actually work and can only use guest from main unit set as router.

Thanks

gnuton, 2022-08-26:
@sirdragonlord AiMesh and guest networks are closed source components the expectation is merlin would work just like stock firmware when it comes to this

gnuton, 2022-08-26:
> @gnuton <https://github.com/gnuton> thanks for the response, so it would
> suggest that merlin would have the same issues with it not working
> correctly across a mesh setup (i'm currently discussing an issue with Asus
> where guest network does not work on 2.4Ghz from a node or if you use
> ethernet backhaul)

Correct. I would expect the same behaviour

idaband, 2022-09-09:
Does this also work with the et8?

jazzyblu, 2022-09-22:
I've read that AiMesh only worked with Ethernet connected (in the beginning?) Is this still the case with the current version? Can I "set up" with Ethernet, disconnect and unplug the satellite and power on somewhere else and have it work via WLAN only? I have a base and two satellites that currently run fine without cable on stock FW.

duncdog, 2023-07-08:
Unsure if this is the place for it, but I don't see it stated anywhere. ASUS began shipping a V2 of this hardware, and explicitly states they need to use their own bespoke versions of the firmware on the main site vs V1 hardware. As this code predates the V2 hardware, I understand this supports V1 hardware. Curious if you could add a clarifier for this hardware sku on the relevant pages for future reference or confusion removal? Thanks!

rshoover2001, 2023-08-27:
Having this precise issue, and bought a third node which happens to be HW V2.

nvram get productid returns "XT8_V2"

The RT-AX95U firmware does not work on this device. Will there be support in the future?

nuller1joe, 2023-08-27:
I just bought another 2 XT8 nodes, however, they are hardware version 2. 
Like @rshoover2001 asked, is there any chance this firmware will support the hw ver v2?
I think the XT8 is more or less useless with stock firmware

jimjim413, 2023-09-10:
I have just brought a XT8 V2 , wondering if I can use Merlin firmware on it ? Thanks

sanjeroHUN, 2023-09-14:
Hi jimjim413! 
I am asking the same question... I have tried to upload the firmware, but nothing happened. Have you tried?

jimjim413, 2023-09-14:
> Hi jimjim413! 
> 
> I am asking the same question... I have tried to upload the firmware, but nothing happened. Have you tried?

Hi ,

I haven't tried it because I believe that V2 using different firmware from V1, so waiting for Merlin firmware for V2 before trying.

gnuton, 2023-09-15:
V2 has different firmware that is build from a different SDK

ehb224, 2023-11-06:
> V2 has different firmware that is build from a different SDK
Does that mean it won't be supported?

gnuton, 2023-11-06:
@ehb224 no it doesn't mean that. It's just a difference which software wise makes the router really different even if the hardware may be similar

ehb224, 2023-11-06:
Any idea on the time frame for v2 firmware? I just bought a pair of these and if I need to return them for a different model I will.  However if we are only looking at a few months I can live with stock firmware. I thank you in advance.

ABitBA, 2023-12-20:
Hey, thanks for your effort to keep the device alive with new firmware and improve it. I only had problems with the V1 as far as a stable backhaul connection is concerned. Hence the change now!

I have installed the firmware 3004.388.4_0-gnuton1 and I noticed that I can only manage 720Mbit via the node with activated backhaul instead of 930Mbit WAN (ASUS official firmware). My computer is connected to the node with 1Gbit and my 1Gbit Internet line could be almost completely exhausted before. On the main router the 930Mbt are fully available... The utilization of the CPU and RAM is not particularly high during the test. The connection between the node and the main router is great

gnuton, 2023-12-20:
@ABitBA I am not sure how you get these values, you can try to reset to factory and try with a fresh installation without previous settings and see if that helps. I am not aware of anything which can cause this.

ABitBA, 2023-12-20:
Thank you for your quick reply. I have reset the router to factory settings and reconfigured it manually. (no old data imported)

Alfiegerner, 2024-02-12:
All those adding comments requesting v2 make sure you add a thumbs up on the requested devices tracker:
 https://github.com/gnuton/asuswrt-merlin.ng/issues/523

haneefmubarak, 2024-07-19:
Hey @gnuton! Sent over an email you about hopefully helping with XT8 v2 support. Just a heads up in case you check GitHub more. Will also join the Discord in case that is easier to chat on. Cheers!