gnuton/asuswrt-merlin.ng #159: Full-cone NAT Support for DSL-AC68U

Randi1115, 2022-03-03:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Can Full-cone NAT option be added as described in the following article to DSL-AC68U please?

[Enabling Full-cone NAT on Asus Wireless Router](https://www.asus.com/support/FAQ/1045254/) 

 As per the article, the latest firmware should support it but I don't see it under WAN options as mentioned. Thank you.

gnuton, 2022-03-04:
Isn't it availale already if you use ethernet as WAN?

adestounis, 2022-03-08:
I'll try it and will be back with any feedback

hiyan, 2022-03-09:
[image]

I have a DSL-AC68U and connected via eth wan (moved away from dsl years back). There is no full-cone nat.

trebleta, 2022-08-01:
This is a way around to enable Full-cone Nat check this post.. untill I fix
https://www.snbforums.com/threads/dsl-ax82u-ax5400-full-cone-nat.80074/

gnuton, 2022-08-08:
Full cone on WAN should be available on DSL-AC68U but Asus has removed it from DSL-AX82U.

c9368920bd5ce47aedb724bc852ff767c2702d1a should add it back to DSL-AX82U

trebleta, 2022-08-08:
Well I've spoken to asus about it, its them who told me the above to swap modes, and they said will prob add it

gnuton, 2022-08-08:
Here is a build with this fix for testing https://output.circle-artifacts.com/output/job/415dcf8d-4106-4e34-830a-f8913f011d20/artifacts/0/workspace/release/DSL-AX82U_386.7_2_cferom_puresqubi.w

trebleta, 2022-09-02:
Is this included in the 386.08, will be testing in a day or two?