gnuton/asuswrt-merlin.ng #160: DoT can't be deleted after deactivated DoT

Hello-olleH, 2022-03-03:
**Router Model Affected**
DSL-AX82U

**Firmware Version Affected**
386.04-gnuton2

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**
DoT can't be deleted after deactivated DoT

Hello-olleH, 2022-03-03:
[image]

You can disable DoT and delete DNS-over-TLS Server List but after save you see DNS-over-TLS Server List again

gnuton, 2022-03-03:
ok clear thanks.. I will take a look..

gnuton, 2022-03-19:
It works on TUF-AX5400 with beta firmware.. I will test the DSL-AX82U too

gnuton, 2022-03-26:
it's a bug. The problem is that you should not see  DoT options when it's disabled ("DNS privacy protocol" is None)