gnuton/asuswrt-merlin.ng #164: TUF-AX5400 386.5 Tools_Sysinfo.asp page menu portal missing

sadoneli, 2022-03-07:
**Router Model Affected**
Models: TUF-AX5400

**Firmware Version Affected**
384.5

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
TUF-AX5400 386.5 Tools_Sysinfo.asp page is avaliable, however menu portal is missing

**Screenshots**
[image]

vladut123, 2022-03-07:
Edit the firmware version is 386.5 not 384

gnuton, 2022-03-13:
ed81a13711 should fix this issue

gnuton, 2022-03-14:
tested but icon missing..

sadoneli, 2022-03-17:
new golden icon for this page maybe needed？

gnuton, 2022-03-19:
can you elaborate more?

gnuton, 2022-03-26:
fixed in the latest stable

closed 2022-03-26 by gnuton