gnuton/asuswrt-merlin.ng #166: Reboot button not work! (and apply buttons)

HamidZaeri, 2022-03-08:
DSL-AC68u - 386.04_0-gnuton2

**Describe the bug**
Reboot button does not reboot the device!

Also in most of GUI sections when no option has changed, apply button not work.
For example in "DSL setting" page if no option is changed and then press apply button a pop-up of "applying settings" appear without the counting percentage and then disappear and DSL connection does not reestablish!

gnuton, 2022-03-09:
Can you please factory reset and try again?
can you also test the latest beta?

HamidZaeri, 2022-03-11:
tried both and did not work.

> Can you please factory reset and try again? can you also test the latest beta?

gnuton, 2022-03-18:
tested the latest beta and reboot and apply work like a charm.

closed 2022-03-24 by gnuton

HamidZaeri, 2022-04-19:
Which device did you test?
I tested on dsl-ac68u and even latest release version 386.5_2 and factory defaults right after update.
and the reboot button (in the GUI) did not work and i have to reboot with the button on back of my router.

> tested the latest beta and reboot and apply work like a charm. I cannot reproduce it.

gnuton, 2022-04-20:
have you tried to flash back the stock firmware?
does the reboot button work for you there?