gnuton/asuswrt-merlin.ng #167: FeatureRequest: Support Firmware upgrading by URL+MD5

tainewoo, 2022-03-09:
**Is your feature request related to a problem? Please describe.**
This feature is not related to any problem, but it would be helpful for users who are remote controlling the rooter.

**Describe the solution you'd like**
Currently for Merlin firmware users, we need to download the released firmware, and upload it to the firmware upgrading page of the administration menu. After the firmware is fully uploaded to the rooter, and extracting and upgrading will be started.
This is the normal behavior and it works well for lan users.
Some other users would have different rooters using merlin firmware, and some of them are not in the same place as they are living, so most of times they will need to control the rooter remotely.
Considering the file size of the firmware, and the stability of the network connecting of the remote controlling, and the timeout limit of connection/response, the remote upgrading will always fail.

For this case, if we can support a new method (just like OTA but not OTA), provide 2 blanks for the administrator to inpurt:
Download URL of the firmware
MD5
After the value is input and applied, the downloading is started (wget?curl?), and after the downloading is completed, the verification of the MD5 would be executed. If everything is fine, the firmware upgrading will started.
All logs (including download, MD5 verification, and everything before the upgrading) can be output to the log interface.

**Describe alternatives you've considered**
Currently there is no other alternatives, rather than login to a device which is in the Lan of the rooter.

vladut123, 2022-03-09:
> **Is your feature request related to a problem? Please describe.** This feature is not related to any problem, but it would be helpful for users who are remote controlling the router.
> 
> **Describe the solution you'd like** Currently for Merlin firmware users, we need to download the released firmware, and upload it to the firmware upgrading page of the administration menu. After the firmware is fully uploaded to the router, and extracting and upgrading will be started. This is the normal behavior and it works well for lan users. Some other users would have different rooters using merlin firmware, and some of them are not in the same place as they are living, so most of times they will need to control the router remotely. Considering the file size of the firmware, and the stability of the network connecting of the remote controlling, and the timeout limit of connection/response, the remote upgrading will always fail.
> 
> For this case, if we can support a new method (just like OTA but not OTA), provide 2 blanks for the administrator to inpurt: Download URL of the firmware MD5 After the value is input and applied, the downloading is started (wget?curl?), and after the downloading is completed, the verification of the MD5 would be executed. If everything is fine, the firmware upgrading will started. All logs (including download, MD5 verification, and everything before the upgrading) can be output to the log interface.
> 
> **Describe alternatives you've considered** Currently there is no other alternatives, rather than login to a device which is in the Lan of the router.

gnuton, 2022-03-09:
Unlikely this will be added in the router itself but it can rather be a custom script (with web ui) which can be
installed with amtm.
AsusWRT do alredy have ability to self-updating themselves to the latest firmware.. so I would rather enable it then having something cutom like this.

tainewoo, 2022-03-09:
> Unlikely this will be added in the router itself but it can rather be a custom script (with web ui) which can be installed with amtm. AsusWRT do alredy have ability to self-updating themselves to the latest firmware.. so I would rather enable it then having something cutom like this.

Thank you for your quick response.
Get it now, so i will pursuit a addon which implements this custom script and web ui instead.
Thank you for your great work on the firmware.

closed 2022-03-13 by gnuton