gnuton/asuswrt-merlin.ng #168: Telnet/SSH options hidden (DSL-AX5400)

seiken27, 2022-03-12:
**Router Model Affected**
Models: DSL-AX5400 (Optus)

**Firmware Version Affected**
386.04

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**
The DSL-AX5400 Optus Version has the telnet/ssh option removed from the administration/system page. 

**Screenshots**
default (currently on beta firmware)
[image]

exposed using a chrome plugin which allowed me to enable SSH and flash your merlin software :D
[image]

DJE007, 2022-03-13:
you can get around this with Chrome Devtools open and passing this on the console

document.getElementById("telnetd_sshd_table").style.display = "";

document.getElementById('sshd_enable_tr').style.display = "";
document.getElementById('sshd_port_tr').style.display = "";
document.getElementById('sshd_password_tr').style.display = "";
document.getElementById('auth_keys_tr').style.display = "";
                
document.form.telnetd_enable[0].disabled = false;
document.form.telnetd_enable[1].disabled = false;
document.form.sshd_enable.disabled = false;

seiken27, 2022-03-14:
yep i used a tool to have it revealed but thought it might be worthwhile to have it exposed for non technical people.

Also not sure if it's possible, but remove any reference to the specific optus model in the firmware as it might be hiding all the AiProtection /adaptive qos which were taken out for this ISP supplied model :(.

gnuton, 2022-03-25:
ae671ed2ef should fix this and it will be part of the next beta

seiken27, 2022-03-25:
awesome! great stuff mate!

gnuton, 2022-03-26:
please feel free to re-open this if something is not right.

closed 2022-03-26 by gnuton

seiken27, 2022-03-27:
yep been fixed in the latest update 👍