gnuton/asuswrt-merlin.ng #172: LED Control Scripts - WiFi LEDS Not Turning back on TUF AX5400

shauno100, 2022-04-02:
**Router Model Affected**
TUF AX-5400

**Firmware Version Affected**
386.5_2-gnuton0

**Is this bug present in upstream Merlin releases too?**
Can't confirm as this router is not officially supported by Merlin

**Describe the bug**
I have enabled scheduled LED control as per https://github.com/RMerl/asuswrt-merlin.ng/wiki/Scheduled-LED-control and when the "/jffs/scripts/ledson.sh" command is run after the LEDs are turned off with the equivalent ledsoff.sh scrip, all the correct LEDs turn back on except the WiFi 2.4Ghz and 5Ghz LEDs. When i reboot the router all the LEDs turn back on. Seems to be a compatibility bug with the LED control system on this particular model and Merlin's "service restart_leds" service maybe?

NOTE: I only use 5Ghz WiFi on my TUF AX5400 so assuming same issue with 2.4Ghz LED but can't be certain as i haven't tested. Definately an issue with 5Ghz LED though.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://github.com/RMerl/asuswrt-merlin.ng/wiki/Scheduled-LED-control and create services-start, ledsoff.sh and ledson.sh scripts as per the instructions.
2. Reboot router
3. SSH into router and run the command with quotes "/jffs/scripts/ledsoff.sh" you will be presented with "Done" message - LEDs are now turned off correctly on router.
4. Now run "/jffs/scripts/ledson.sh" and notice the WAN, LAN and Power LEDs turn back on successfully but the WiFi 5Ghz and 2.4Ghz LEDs remain off.

**Expected behavior**
WiFi 2.4Ghz/5Ghz to turn back on as intended

trebleta, 2022-07-05:
I get something like this on the DSL AX82U. If i disable the leds via the web UI then try to enable nothing comes back on. Same in lastest beta 2 firmware.

trebleta, 2022-07-05:
No he has said it's a problem his trying to fix

trebleta, 2022-07-07:
Quick question if you use the asus mobile app can you turn on off etc, for me on my dsl ax82u that works. Yet via any other way they will not come on including trying to turn back on via the app if the all did not turn it off

gnuton, 2022-07-09:
@trebleta this was an issue of 386.5 see the changelog 

https://github.com/gnuton/asuswrt-merlin.ng/blob/gnuton-master/Changelog-NG.txt

gnuton, 2022-07-09:
@shauno100 this should have  been fixed. if you have time please check the latest beta.

gnuton, 2022-07-09:
checked the issue is still there.

trebleta, 2022-07-11:
Maybe try to use the same way the mobile app controls the leds

gnuton, 2022-07-11:
@trebleta I have investigated the issue and I am working on a fix.

shauno100, 2022-07-11:
@gnuton thanks for looking into the issue, unfortunately I am no longer using the TUF AX5400 router so cannot test. Hopefully you're able to get the LED issue fixed though 🙂

gnuton, 2022-07-11:
Sure no problem at all. Thanks for reporting it

trebleta, 2022-07-19:
Just wanted to say when I was saying leds I did mean the RGB

gnuton, 2022-07-28:
TUF-AX5400 should be fixed in the last beta4, but  DSL-AX82U had not been fixed yet.

trebleta, 2022-09-02:
Any update on this?

gnuton, 2022-09-03:
Nn updates about the dsl-ac82u. I tried to fix it didn't work the code is much compels there and i may have missed something.

trebleta, 2022-09-03:
Well on the offical you can completely control the RGB on the main web ui page, maybe just copy and add that part to your merlin.
[image]

trebleta, 2022-09-03:
And look how the offical turns it off and on and add to the led/ rgb control script

trebleta, 2022-09-03:
I am a coder, just old skool basic, c++ pascal

gnuton, 2022-09-03:
That part works. You can switch on and off aura. The part which does not
work is switching on again the 4 LEDs on the left side.
The code which handles the gpios for the LEDs is C so you should familiar
with the code. :)

trebleta, 2022-09-03:
If you send me that bit of code, I should be able to yes, 
before it was turning the RGB off and not back on so that side is now fixed?

trebleta, 2022-09-03:
Am sorry I have not tried you firmware since I had dsl problems. I'm still having 1-2 problems, also 1-2 issues I'm reporting to asus at mo. 1 to do with wifi, other DDNS.

gnuton, 2022-09-03:
No worries. Using stock firmware is not a crime ;) i will update this when this is issue gets fixed. Have a nice weekend