gnuton/asuswrt-merlin.ng #179: Asus XT8 - Protocol 8100 is buggy, dev wds2.0.1

stebrick, 2022-04-19:
**ASUS ZenWiFi XT8**
Models: (eg: DSL-AX82U/RT-AX82U/...)

**Firmware Version Affected**
386.5_2-gnuton0

**Is this bug present in upstream Merlin releases too?**
I am not sure about this.

**Describe the bug**

"_kernel: protocol 8100 is buggy, dev wds2.0.1_"

My router randomly disconnects my devices.  I found this abnormality in my router logs and suspect this is one of the possible causes, tho this log doesn't appear right before the disconnection.  

There's another recent report of this buggy thing here: https://www.snbforums.com/threads/may-5-13-05-14-crashlog-6-protocol-8100-is-buggy-dev-wds2-0-1.78175/

gnuton, 2022-04-20:
@stebrick thanks for reporting this.
Let's see if this is happening on stock firmware too.
I will be following the thread.

stebrick, 2022-04-21:
> @stebrick thanks for reporting this. Let's see if this is happening on stock firmware too. I will be following the thread.

Thank you.  I can confirm that I cant find this particular error line in my system log anymore ever since my router reverts to stock firmware.   And those network drop out I mentioned before still persisted in stock firmware and might not be related to this error.  It seems more likely to be an ISP issue with DNS setting. 

Weird tho, I've never experienced router crash in your latest custom firmware.  Yet there's another fresh report in the link I posted before.  Just FYI.