gnuton/asuswrt-merlin.ng #181: Support for TUF-AX3000

gnuton, 2022-04-25:
This thread is about supporting TUF-AX3000 
Feel free to ask questions here. 

Status:
- [] received GPL
- [] received development unit
- [] Merged GPL to feature-branch
- [] Initial testing & Bugfixing phase
- [] Public pre-alphas
- [] Alpha released
- [] collecting testing reports  & Bugfixing phase
- [] Beta released
- [] collecting testing reports  & Bugfixing phase
- [] Stable released

ETAs:
alpha: TBD
beta: TBD
stable: TBD (387.x)

qscface, 2022-04-27:
Thanks mate! Looking forward to your great work!

yk00365, 2022-04-27:
Thanks a lot！

listen844, 2022-04-27:
I look forward to this day very much!

Jacky10X, 2022-05-01:
Does this firmware support assassin mode ? It seems this feature only available China version (Not the Asia version one).

MaDaLiNoSt, 2022-05-02:
If lan4 can be used as WAN in dual wan mode (LAN4+WAN PORT)
can be dual wan mode reconfigured to use LAN1+LAN4 and the 2.5G port as LAN port?

This could be usefull in Aggregation mode (1G+1G) to get 2G out via 2.5G out via WAN port instead of only relying on wifi.

Can this be achieved?
Thanks!

betiau, 2022-05-08:
Thank you for picking this up! Looking forward to this a lot! :)

Ferdze, 2022-05-09:
Niiiice!!! I It will be great for the tuf ax3000 to have a custom firmware

junskychn, 2022-05-14:
Thanks mate! Looking forward to your great work!

gnuton, 2022-05-15:
@MaDaLiNoSt google to find a way to do that. it's in the merlin docs
As update we are still waiting for the new GPLs from ASUS.

Rovasteen, 2022-05-31:
The GS-AX3000 is yet another variant of this hardware platform with a different case.  Is this router supported as part of the work on the TUF-AX3000 model?

-------------------------------------------------------------------

Asus site:   https://rog.asus.com/us/networking/rog-strix-gs-ax3000-model/helpdesk_bios

Current firmware:  https://dlcdnets.asus.com/pub/ASUS/wireless/GS-AX3000/FW_GS_AX3000_300438643588.zip

Version 3.0.0.4.386.43588
2021/08/17 51.82 MBytes
ASUS ROG STRIX GS-AX3000 Firmware version 3.0.0.4.386.43588
1. Fix AiMesh issues.
2. Improve system stability.
3. Fix WAN DNS setting cannot setup LAN side pihole server.
4. Modify few IPTV settings.
5. Add Google assistant supp

fan881217, 2022-06-02:
My router is TUF-AX3000，Fireware: 3.0.0.4.386_41700_koolshare. After a period of normal access, the Xiaomi camera is disconnected from the network. However, the camera can be seen from the router, but the intranet cannot be accessed. The Mijia APP shows that the camera is offline. After restarting the router, the network connection can be restored

gnuton, 2022-06-05:
First alpha  available FOR TESTERS ONLY.
This image is not safe for everyday use.

https://output.circle-artifacts.com/output/job/96d2b641-8439-4a54-bfd4-bd6375d43fb9/artifacts/0/workspace/release/TUF-AX3000_386.7_alpha1_cferom_puresqubi.w
MD5sum
91f9b54e11b45fc684a365d5d0db5704  /home/docker/project/release/src-rt-5.02axhnd.675x/targets/96750GW/TUF-AX3000_386.7_alpha1_cferom_puresqubi.w

Rovasteen, 2022-06-06:
Awesome.  That is very cool.

yk00365, 2022-06-06:
Awesome.

gnuton, 2022-06-06:
please test it and share feedback

yk00365, 2022-06-06:
> please test it and share feedback

Sure！That is very cool.

Rovasteen, 2022-06-06:
Looking forward to Asus providing the GPL to Gnuton for the GS-AX3000.  I would certainly test an alpha version of that!

JPatrick2398, 2022-06-07:
[image]

Issue: Unable to make changes on 2.4G, I tried to change the channel bandwidth and control channel but the changes won't apply even after hitting the apply button, but it is still working and I can connect to 2.4ghz without issues. 5Ghz tab seems to be working fine though.

Suggestion: In the network map tab can you also set the system status as the default view instead of the wireless settings just like merlin's.

I will test the vpn using different protocols next.

It seems like a great build not a lot of issues, other functions and features are working the way it should. In fact I might flash this build to my 2nd TUF-AX3000 router for everyday use, because it is stable enough for me already.

Thank you and great work Gnuton!

yk00365, 2022-06-08:
> [image]
> 
> Issue 1: Unable to make changes on 2.4G, I tried to change the channel bandwidth and control channel but the changes won't apply even after hitting the apply button, but it is still working and I can connect to 2.4ghz without issues. 5Ghz tab seems to be working fine though.
> 
> Issue 2: Wifi Radar tab is also not working "http://192.168.50.1/WiFi_Insight.asp"
> 
> Suggestion: In the network map tab can you also set the system status as the default view instead of the wireless settings just like merlin's.
> 
> I will test the vpn using different protocols next.
> 
> It seems like a great build not a lot of issues, other functions and features are working the way it should. In fact I might flash this build to my 2nd TUF-AX3000 router for everyday use, because it is stable enough for me already.
> 
> Thank you and great work Gnuton!

Yes,the bug is all the same ,and thank you and great work Gnuton!

Ferdze, 2022-06-08:
does it have a working fq-codel? thanks!

JPatrick2398, 2022-06-08:
> does it have a working fq-codel? thanks!

No, it was removed by merlin on build 386.1. (https://www.asuswrt-merlin.net/changelog)

REMOVED: fq_codel support for Adaptive QoS.  Due to a change
             in how Trend Micro configures QoS, it is no longer
             possible to intercept these to inject fq_codel.

yk00365, 2022-06-08:
> does it have a working fq-codel? thanks!

No，

CrashX, 2022-06-16:
TUF-AX3000 & GS-AX5400

gnuton, 2022-06-20:
GS ax5400 needs different GPL and support. No plans to support it yet.

yk00365, 2022-06-20:
> Hello, does this tuf-ax3000 firmware support assassin mode?

I using that,it is supported,this mode only rely on the cfe firmware. Only in China Mainland.

Partizanin, 2022-06-23:
Issue with vpn.
Vpn do not work properly, after import file and apply all configuration, when i toggle on vpn service with "service state" toggle button it should change my local dns settings but it don`t.
Some screenshots and vpn file are below.

<details>
<summary>Here some screenshots with samples</summary>
<br>

[vpn config file](https://zaborona.help/zaborona-help.ovpn)

[image]

[image]

</details>

qscface, 2022-07-05:
thx for your great work

tzgroute, 2022-07-06:
xd6 for meilin!! thanks!

Partizanin, 2022-07-07:
System time is not synchronized with ntp server after reboot

gnuton, 2022-07-07:
@Partizanin please file this issues in a separate issue so that I can track them better otherwise they get forgotten. thanks

Jacky10X, 2022-07-19:
I suggest to change it to TUF-AX3000 v1 since there are two models (v1 & v2). 
Some people might conflict with these 2 models (they are different specs).

yk00365, 2022-07-20:
> I suggest to change it to TUF-AX3000 v1 since there are two models (v1 & v2). Some people might conflict with these 2 models (they are different specs).

Hi，the tuf-ax3000v1's CPU model is not same as V2's, V2 is a new model, the author also waiting for ASUS the GPL source code for the V2, the firmware core is not same as V1's.So please vote thumb up at https://github.com/gnuton/asuswrt-merlin.ng/issues/140. Thanks.

hm-tamim, 2022-07-25:
Is the stable version released?

Partizanin, 2022-07-26:
> Is the stable version released?

Not yet, follow of this page https://github.com/gnuton/asuswrt-merlin.ng/releases

gnuton, 2022-08-07:
Stable released!

gnuton, 2022-08-15:
Since the stable has been release I will be closing this ticket.

closed 2022-08-15 by gnuton