gnuton/asuswrt-merlin.ng #182: Excessive dns.msftncsi.com dns queries

denis-g, 2022-04-27:
**Router Model Affected**
Models: RT-AX82U

**Firmware Version Affected**
386.5_2-gnuton0

**Is this bug present in upstream Merlin releases too?**
Maybe [1](https://www.snbforums.com/threads/dns-requests-for-dns-msftncsi-com.76700/) [2](https://4pda.to/pages/go/?u=https%3A%2F%2Fwww.snbforums.com%2Fthreads%2Fconstant-unwanted-traffic-to-dns-msftncsi-com-from-rt-ac66u.35367%2F&e=114529982)

**Describe the bug**
Excessive dns.msftncsi.com dns queries, every 15-20 sec.
"Auto Network Detection" is disabled, also "DNS Query" set is empty.

**To Reproduce**
Steps to reproduce the behavior:
1. Install fw
2. Reset config
3. Set DNS to Pihole address
4. See Query log

**Expected behavior**
On vanilla fw 3.0.0.4.386.46061 don't have this problem.

**Screenshots**
[image]

gnuton, 2022-04-29:
have you tried to disable this from http://router.asus.com/Advanced_System_Content.asp

[image]

gnuton, 2022-04-29:
Hi!
try to set dns_norebind to 0 in the nvram.
It looks like stubby configuration is using it by default.

axeloz, 2022-05-01:
Hello, 

I had this issue on genuine Asus firmware. I switched today to Merlin and this issue persists, same as @denis-g.

Probe is disabled in the settings but the router keeps resolving dns.msftncsi.com (I know this is the router because I have NO Windows computer at home):
[image]

As you advised, I tried to 
```
nvram set dns_norebind=0
nvram commit
reboot
```

But it keeps calling. I also have: 
```
axel@ZenWiFi_XT8-EA80:/tmp/home/root# nvram show |grep prob
dns_probe=0
dns_probe_content=131.107.255.255 112.4.20.71 fd3e:4f5a:5b81::1
dns_probe_host=dns.msftncsi.com
```

`dns_probe` is supposed to be off. 
It seems Merlin has pushed a fix for this: 
https://github.com/RMerl/asuswrt-merlin.ng/commit/8943404b78901f32954d93c4e27a92c998b725c6
But this is quite old.

Thanks

gnuton, 2022-05-01:
do you see dns.msftncsi.com in /etc/stubby/stubby.yml ?

axeloz, 2022-05-02:
```
axel@ZenWiFi_XT8-EA80:/tmp/home/root# cat  /etc/stubby/stubby.yml 
cat: can't open '/etc/stubby/stubby.yml': No such file or directory
```

gizmosngadgetz, 2022-05-03:
I cannot find a 'stubby.yml' file in my router. 

When dns.msftnsci.com is blocked w/ pihole, my Asus TUF-AX5400 says its "disconnected" from the internet even though there is still actually a connection. The LED on the front panel of the router that indicates online status also turns RED (Unblocking dns.msftnsci.com turns the LED back to white). Another side effect of blocking dns.msftnsci.com with pihole is Windows also displays "no internet connection" (in the system icon tray) when connected via wifi to the router (but similarily, there is still connection to the internet).

May I speculate that it's Asus' core firmware that is using dns.msftnsci.com to check for online status? 

I keep dns.msftnsci.com blocked, turned on stealth mode for the router (all LEDs off), and hidden the system icon tray in windows.

gnuton, 2022-05-03:
@gizmosngadgetz correct that's what  dns.ismsftnsci.com is for. and if you stop it you must use a ping or something to make the router understand that it's online. This is not a bug, it's intended to work like this.
Blocking it you will also have some other issues in the router like dual wan not working.

diegocr, 2024-08-25:
On stock `3.0.0.6.102_34314-g8ae1379_400-g562af` the interval seems to be 60 seconds, still i do find that too short... i would set something like 10 minutes and it should be fine.

Really wondering why isn't there an option (either web, or nvram) to control this interval 🙄

rochmind, 2025-01-04:
> Hello,
> 
> I had this issue on genuine Asus firmware. I switched today to Merlin and this issue persists, same as @denis-g.
> 
> Probe is disabled in the settings but the router keeps resolving dns.msftncsi.com (I know this is the router because I have NO Windows computer at home): [image]
> 
> As you advised, I tried to
> 
> ```
> nvram set dns_norebind=0
> nvram commit
> reboot
> ```
> 
> But it keeps calling. I also have:
> 
> ```
> axel@ZenWiFi_XT8-EA80:/tmp/home/root# nvram show |grep prob
> dns_probe=0
> dns_probe_content=131.107.255.255 112.4.20.71 fd3e:4f5a:5b81::1
> dns_probe_host=dns.msftncsi.com
> ```
> 
> `dns_probe` is supposed to be off. It seems Merlin has pushed a fix for this: [RMerl/asuswrt-merlin.ng@8943404](https://github.com/RMerl/asuswrt-merlin.ng/commit/8943404b78901f32954d93c4e27a92c998b725c6) But this is quite old.
> 
> Thanks
> 
> EDIT : FYI I'm using NextDNS.com as DNS provider. The free subscription allows 300 000 requests/mo which is huge but not enough in my case: 1/4 or 1/3 of all the DNS lookups I make are for dns.msftncsi.com, which leads to exceed this quota.

I’ve setup pihole with nextdns as upstream nameserver, the caching helps 2/3 of the calls. also as a hack, i’ve setup 2 free nextdns accounts and pihole points to both 😁 (effectively doubles quota)

I have this issue on my new aimesh node (rt-ax92u) but it does not seem to be a problem on the main router. (rt-ax92u)