gnuton/asuswrt-merlin.ng #184: Select ISP Profile is not shown

feeler29, 2022-05-06:
The "Select ISP Profile" is not shown on IPTV page . 

The router is Optus AX5400.  I think Optus has hidden this option.  

After flashing  "DSL-AX82U_386.5_2-gnuton0_cferom_puresqubi.w", it still not seen.

Would you kindly consider to fix this on next revision? Thank you.

[image]

gnuton, 2022-05-06:
I will take a look at this

codebullfrog, 2022-10-07:
Option not available on DSL AC68U either. I has had to resort the instructions [here](https://community.tpg.com.au/t5/Modems-and-Devices/ASUS-DSL-AC68U-on-TPG-HFC/td-p/7762/page/5) to get it working with my ISP (not tpg).