gnuton/asuswrt-merlin.ng #185: DNS leak on VPN client

quaningeneva, 2022-05-08:
**Router Model Affected**
Models: AX-XT8

**Firmware Version Affected**
386.5_2

**Is this bug present in upstream Merlin releases too?**
I don't know.

**Describe the bug**
When doing the DNS leak test (www.dnsleaktest.com), the IP address correctly comes from the VPN but the DNS server shows the local server and not the VPN server.

**To Reproduce**

**Expected behavior**
VPN connection should show DNS server of VPN and not of local router.

**Screenshots**

quaningeneva, 2022-06-04:
I am not running the DNS behind a VPN.  When I use OpenVPN on the device itself, I don’t get a leak.

quaningeneva, 2022-06-04:
> Force it be behind a vpn it'll fix the leak.

This defeats the purpose of having the VPN client and routing on the router.  I am hoping that this DNS leak can be fixed in a future firmware release.

trebleta, 2022-07-02:
It also depends on the vpn rule for dns needs to be exclusive, if you are not using a vpn for all, else strict will leak your dns etc