gnuton/asuswrt-merlin.ng #186: Link Aggregation on ZenWifi XT8

TBONE204, 2022-05-13:
Is it possible to develop the ability for link aggregation on the XT8 through the firmware?  
It would be a very useful feature especially for those who utilize a NAS or other devices where bandwidth  could become a concern due to large file transfer, many users, or media library streaming.

I'm unclear why the official firmware doesn't allow this.  If this feature would be a hardware limitation I understand.

thanks

gnuton, 2022-05-13:
Please have a look at https://github.com/RMerl/asuswrt-merlin.ng/wiki/Link-Aggregation

closed 2022-05-17 by gnuton

ChloeMarieTaylor76, 2022-09-18:
Hi, I also have the xt8 and would greatly appreciate link aggregation enabled in the firmware. I tried the guide above and added the file and script but it is not working for me. I was trying to set up lacp which my switch supports but it was stopping dchp from working. It mentions supporting two ports but doesn't say which, I tried the last two 2 and 3 with no success. Is there plans to add it please?