gnuton/asuswrt-merlin.ng #191: Unable to save WAN configuration with DoT enabled

ChargedByte, 2022-05-25:
**Router Model Affected**
Models: DSL-AX82U

**Firmware Version Affected**
386.5_2-gnuton0

**Is this bug present in upstream Merlin releases too?**
Not that I could find.

**Describe the bug**
WAN configuration cannot be saved if DNS-over-TLS (DoT) is enabled, also adding items to the DoT Server List does not work.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to WAN settings
2. Select DoT from the DNS Privacy Protocol list
3. Type details into the DoT Server List and click on the "plus" sign, nothing happens (no newline)
4. Scroll to the bottom and click "Ok", nothing happens

**Expected behaviour**
Expected to be able to enable DoT, add servers to list and save.

**Screenshots**
[image]

[image]

gnuton, 2022-05-26:
Hi,
I have just tested this on my router and all works fine. Please reset to factory.
Here is the proof
[image]

ChargedByte, 2022-05-27:
I did just factory reset the device and tried again but the issue remains.

gnuton, 2022-05-27:
Interesting.. 
We are getting new GPLs. I will try to get it fixed in the next release. I will keep this ticket open

gnuton, 2022-06-27:
This should be fixed now in the beta, does anybody confirm that?

ChargedByte, 2022-06-28:
I can confirm. This is fixed in 386.07_0-gnuton0_beta1.

There was a malformed entry in the list (see image), but hitting remove on it worked, and it seems to have gone away for good. I also checked with Cloudflare's Connection Information (https://1.1.1.1/help) and everything was working as expected.

[image]

closed 2022-07-05 by gnuton