gnuton/asuswrt-merlin.ng #200: GUI issues with new alpha code

SkrubbinBubb1es, 2022-06-17:
**Router Model Affected**
Models: TUF5400

**Firmware Version Affected**
[386.07_0-gnuton0_alpha0]

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
under general wifi settings, 2.4Ghz section, the gui is messed up. the 5ghz section is fine though. does not affect anything as my settings from before the upgrade are still good. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

GaselK, 2022-06-18:
+1 for my TUF AX5400. Same behavior.

gnuton, 2022-06-26:
Thakns for reporting this. just tested the new beta and this has been fixed. Doing some verifications then it willbe available

SkrubbinBubb1es, 2022-06-27:
Just loaded the new beta and I can confirm that this issue is resolved, thank you.

closed 2022-06-27 by SkrubbinBubb1es