gnuton/asuswrt-merlin.ng #201: WAN - Internet Connection -  doesn´t show option : pvc edit and add/delete

ICIOMAGO, 2022-06-18:
**Router Model Affected**
DSL-AX82U

**Firmware Version Affected**
386.5_2-gnuton0

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
Doesn´t show wan option : pvc edit and add/delete

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'wan'
2. Click on 'internet connection....'
3. See error

**Expected behavior**
Doesn't show wan option pvc edit and add/delete

**Screenshots**
[image]

gnuton, 2022-06-26:
That's a stable version.
This issue should disappear if you reset to factory

gnuton, 2022-06-29:
[image]
this seems to be fine also with the latest pre-release for me.
closing the ticket

closed 2022-06-29 by gnuton