gnuton/asuswrt-merlin.ng #202: Repetive log entries

GaselK, 2022-06-27:
**Router Model Affected**
Models: TUF AX5400

**Firmware Version Affected**
TUF-AX5400_386.07_0-gnuton0_beta1

**Describe the bug**
Repetive log entries filling log

[image]

**To Reproduce**
System Log tab

gnuton, 2022-06-27:
can you show me more the logs? I cannot see anything wrong in the snippet you shown me

GaselK, 2022-06-27:
Jun 27 23:51:12 rc_service: watchdog 1886:notify_rc stop_aae
Jun 27 23:51:12 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 27 23:51:12 rc_service: waitting "stop_aae" via watchdog ...
Jun 27 23:51:43 rc_service: watchdog 1886:notify_rc stop_aae
Jun 27 23:51:43 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 27 23:51:43 rc_service: waitting "stop_aae" via watchdog ...
Jun 27 23:52:14 rc_service: watchdog 1886:notify_rc stop_aae
Jun 27 23:52:14 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 27 23:52:14 rc_service: waitting "stop_aae" via watchdog ...
Jun 27 23:52:45 rc_service: watchdog 1886:notify_rc stop_aae
Jun 27 23:52:45 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 27 23:52:45 rc_service: waitting "stop_aae" via watchdog ...
Jun 27 23:53:16 rc_service: watchdog 1886:notify_rc stop_aae
Jun 27 23:53:16 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 27 23:53:16 rc_service: waitting "stop_aae" via watchdog ...
Jun 27 23:53:47 rc_service: watchdog 1886:notify_rc stop_aae
Jun 27 23:53:47 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 27 23:53:47 rc_service: waitting "stop_aae" via watchdog ...
Jun 27 23:54:18 rc_service: watchdog 1886:notify_rc stop_aae
Jun 27 23:54:18 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 27 23:54:18 rc_service: waitting "stop_aae" via watchdog ...
Jun 27 23:54:49 rc_service: watchdog 1886:notify_rc stop_aae
Jun 27 23:54:49 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 27 23:54:49 rc_service: waitting "stop_aae" via watchdog ...
Jun 27 23:55:20 rc_service: watchdog 1886:notify_rc stop_aae
Jun 27 23:55:20 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 27 23:55:20 rc_service: waitting "stop_aae" via watchdog ...
Jun 27 23:55:51 rc_service: watchdog 1886:notify_rc stop_aae
Jun 27 23:55:51 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 27 23:55:51 rc_service: waitting "stop_aae" via watchdog ...
Jun 27 23:56:22 rc_service: watchdog 1886:notify_rc stop_aae
Jun 27 23:56:22 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 27 23:56:22 rc_service: waitting "stop_aae" via watchdog ...
Jun 27 23:56:53 rc_service: watchdog 1886:notify_rc stop_aae
Jun 27 23:56:53 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 27 23:56:53 rc_service: waitting "stop_aae" via watchdog ...
Jun 27 23:57:24 rc_service: watchdog 1886:notify_rc stop_aae
Jun 27 23:57:24 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 27 23:57:24 rc_service: waitting "stop_aae" via watchdog ...
Jun 27 23:57:55 rc_service: watchdog 1886:notify_rc stop_aae
Jun 27 23:57:55 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 27 23:57:55 rc_service: waitting "stop_aae" via watchdog ...
Jun 27 23:58:26 rc_service: watchdog 1886:notify_rc stop_aae
Jun 27 23:58:26 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 27 23:58:26 rc_service: waitting "stop_aae" via watchdog ...
Jun 27 23:58:57 rc_service: watchdog 1886:notify_rc stop_aae
Jun 27 23:58:57 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 27 23:58:57 rc_service: waitting "stop_aae" via watchdog ...
Jun 27 23:59:28 rc_service: watchdog 1886:notify_rc stop_aae
Jun 27 23:59:28 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 27 23:59:28 rc_service: waitting "stop_aae" via watchdog ...
Jun 27 23:59:59 rc_service: watchdog 1886:notify_rc stop_aae
Jun 27 23:59:59 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 27 23:59:59 rc_service: waitting "stop_aae" via watchdog ...
Jun 28 00:00:30 rc_service: watchdog 1886:notify_rc stop_aae
Jun 28 00:00:30 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 28 00:00:30 rc_service: waitting "stop_aae" via watchdog ...
Jun 28 00:01:01 rc_service: watchdog 1886:notify_rc stop_aae
Jun 28 00:01:01 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 28 00:01:01 rc_service: waitting "stop_aae" via watchdog ...
Jun 28 00:01:32 rc_service: watchdog 1886:notify_rc stop_aae
Jun 28 00:01:32 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 28 00:01:32 rc_service: waitting "stop_aae" via watchdog ...
Jun 28 00:02:03 rc_service: watchdog 1886:notify_rc stop_aae
Jun 28 00:02:03 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 28 00:02:03 rc_service: waitting "stop_aae" via watchdog ...
Jun 28 00:02:34 rc_service: watchdog 1886:notify_rc stop_aae
Jun 28 00:02:34 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 28 00:02:34 rc_service: waitting "stop_aae" via watchdog ...
Jun 28 00:03:05 rc_service: watchdog 1886:notify_rc stop_aae
[... 34 lines cut ...]
Jun 28 00:08:46 rc_service: waitting "stop_aae" via watchdog ...
Jun 28 00:09:17 rc_service: watchdog 1886:notify_rc stop_aae
Jun 28 00:09:17 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 28 00:09:17 rc_service: waitting "stop_aae" via watchdog ...
Jun 28 00:09:48 rc_service: watchdog 1886:notify_rc stop_aae
Jun 28 00:09:48 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 28 00:09:48 rc_service: waitting "stop_aae" via watchdog ...
Jun 28 00:10:19 rc_service: watchdog 1886:notify_rc stop_aae
Jun 28 00:10:19 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 28 00:10:19 rc_service: waitting "stop_aae" via watchdog ...
Jun 28 00:10:50 rc_service: watchdog 1886:notify_rc stop_aae
Jun 28 00:10:50 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 28 00:10:50 rc_service: waitting "stop_aae" via watchdog ...
Jun 28 00:11:21 rc_service: watchdog 1886:notify_rc stop_aae
Jun 28 00:11:21 rc_service: watchdog 1886:notify_rc start_mastiff
[... cut ...]
Jun 28 00:11:52 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 28 00:11:52 rc_service: waitting "stop_aae" via watchdog ...
Jun 28 00:12:23 rc_service: watchdog 1886:notify_rc stop_aae
Jun 28 00:12:23 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 28 00:12:23 rc_service: waitting "stop_aae" via watchdog ...
Jun 28 00:12:54 rc_service: watchdog 1886:notify_rc stop_aae
Jun 28 00:12:54 rc_service: watchdog 1886:notify_rc start_mastiff
Jun 28 00:12:54 rc_service: waitting "stop_aae" via watchdog ...

JYYJ1986, 2022-06-28:
I’m also having the same log spam.. Already factory reseted..
[image]

GaselK, 2022-06-28:
> 
Yes, router was factory reset after fw install. Settings by hand. JFFS is not enabled.

gnuton, 2022-06-29:
do you have a particular setup in place? AIMesh or other?

GaselK, 2022-06-29:
No particular setup.No AiMesh. Only a 2 DNS-over-TLS Servers (adguard at the moment).

JYYJ1986, 2022-06-30:
Sorry.. I’m on AX82U (not dsl model). I’m also using DOT server (nextdns) on both ipv4/6.

gnuton, 2022-06-30:
ok I managed to reproduce the issue and I think I found the problem
The problem is a missing library which doesn't allow mastiff and aawes to start

gnuton@TUF-AX5400-F468:/tmp/home/root# mastiff 
mastiff: error while loading shared libraries: libssh2.so.1: cannot open shared object file: No such file or directory
gnuton@TUF-AX5400-F468:/tmp/home/root# aaews 
aaews: error while loading shared libraries: libssh2.so.1: cannot open shared object file: No such file or directory

Karmakstylez, 2022-07-01:
I have the same issue. Will this be updated/fixed in the next beta?

gnuton, 2022-07-02:
@Karmakstylez yes I fixed this and tested it.
I will most probably release the beta2 today.

GaselK, 2022-07-02:
Fixed on new beta 2 !

closed 2022-07-02 by GaselK