gnuton/asuswrt-merlin.ng #204: 386.07_0-gnuton0_beta1 DHCP not working on the WAN link

DJE007, 2022-06-29:
**Router Model Affected**
DSL-AX82U

**Firmware Version Affected**
386.07_0-gnuton0_beta1 

Wan link DSL comes up but DHCP never gets an IP from the ISP provider, reverted to older Stable: 386.5_2-gnuton0 and works again.

gnuton, 2022-06-30:
which ISP do you have?
I am running 386.7 on my DSL-AX82U and I have not seen this issue yet.

gnuton, 2022-06-30:
Since I am not able to reproduce this issue yet, I mark it as support ticket.
I will remove that label if I do find this is an actual issue with some ISP

DJE007, 2022-07-01:
[syslog.txt](https://github.com/gnuton/asuswrt-merlin.ng/files/9026852/syslog.txt)

DJE007, 2022-07-01:
May  5 15:05:22 kernel: klogd started: BusyBox v1.24.1 (2022-03-25 11:55:14 CST)
May  5 15:05:22 kernel: [    0.000000] Linux version 4.1.52 (sam@AA1100887-PLX1) (gcc version 5.5.0 (Buildroot 2017.11.1) ) #2 SMP PREEMPT Fri Mar 25 12:39:31 CST 2022
May  5 15:05:22 kernel: [    0.000000] Kernel command line: root=/dev/ubiblock0_0 ubi.mtd=0 ubi.block=0,0 rootfstype=squashfs console=ttyAMA0 earlyprintk debug irqaffinity=0 pci=pcie_bus_safe
May  5 15:05:22 kernel: [    0.000000] Virtual kernel memory layout:
May  5 15:05:22 kernel: [    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
May  5 15:05:22 kernel: [    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
May  5 15:05:22 kernel: [    0.000000]     vmalloc : 0xe0800000 - 0xff000000   ( 488 MB)
May  5 15:05:22 kernel: [    0.000000]     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
May  5 15:05:22 kernel: [    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
May  5 15:05:22 kernel: [    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
May  5 15:05:22 kernel: [    0.000000]       .text : 0xc0018000 - 0xc05a5d60   (5688 kB)
May  5 15:05:22 kernel: [    0.000000]       .init : 0xc05a6000 - 0xc05da000   ( 208 kB)
May  5 15:05:22 kernel: [    0.000000]       .data : 0xc05da000 - 0xc060b360   ( 197 kB)
May  5 15:05:22 kernel: [    0.000000]        .bss : 0xc060b360 - 0xc0671c5c   ( 411 kB)
May  5 15:05:22 kernel: [    0.414719]    adsl   0x0fe00000  0xcfe00000  0x00200000
May  5 15:05:22 kernel: [    0.420202] CCI hardware cache coherency enabled
May  5 15:05:22 kernel: [    0.424980] pmc_init:PMC using DQM mode
May  5 15:05:22 kernel: [    0.428978] Successfully retrieved NVRAM data from dtb
May  5 15:05:22 kernel: [    0.434288] cfe-v^A: 1.0.38-164.255
May  5 15:05:22 kernel: [    0.556242] Broadcom Logger v0.1
May  5 15:05:22 kernel: [    0.567454] skb_free_task created successfully with start budget 256
May  5 15:05:22 kernel: [    0.573949] ^[[0;34mBLOG v3.0 Initialized^[[0m
May  5 15:05:22 kernel: [    0.578229] BLOG Rule v1.0 Initialized
May  5 15:05:22 kernel: [    0.581978] Broadcom IQoS v1.0 initialized
May  5 15:05:22 kernel: [    0.586072] Broadcom GBPM v0.1 initialized
May  5 15:05:22 kernel: [    0.693582] SPI NAND Device Linux Registration
May  5 15:05:22 kernel: [    0.698126] SPI NAND Linux Registration
May  5 15:05:22 kernel: [    0.702266] SPI NAND device reset
May  5 15:05:22 kernel: [    0.715957] SPI NAND device Micron MT29F2G01AB
May  5 15:05:22 kernel: [    0.720404]    device id    = 0x2c242c24
May  5 15:05:22 kernel: [    0.724349]    ecc          = 6/8 bits
May  5 15:05:22 kernel: [    0.728097]    page size    = 0x800 (2048) bytes
May  5 15:05:22 kernel: [    0.732715]    block size   = 0x20000 (131072) bytes
May  5 15:05:22 kernel: [    0.737680]    total blocks = 0x800 (2048)
May  5 15:05:22 kernel: [    0.741777]    total size   = 0x10000000 (268435456) bytes
May  5 15:05:22 kernel: [    0.869747] >>>>> For primary mtd partition rootfs, cferam/vmlinux.lz UBI volume, vmlinux fs mounted as squash fs on UBI <<<<<
May  5 15:05:22 kernel: [    0.900257] Secondary mtd partition rootfs_update detected as UBI for cferam/vmlinux source and squash fs on UBI for vmlinux filesystem
May  5 15:05:22 kernel: [    0.912450] setup_mtd_parts: misc indx 2 name misc3 nvram configured size 1 
May  5 15:05:22 kernel: [    0.919505] setup_mtd_parts: name misc3 configured size 0x00100000 offset 0xF600000
May  5 15:05:22 kernel: [    0.927167] setup_mtd_parts: misc indx 1 name misc2 nvram configured size 47 
May  5 15:05:22 kernel: [    0.934306] setup_mtd_parts: name misc2 configured size 0x02f00000 offset 0xC700000
May  5 15:05:22 kernel: [    0.941966] setup_mtd_parts: misc indx 0 name misc1 nvram configured size 8 
May  5 15:05:22 kernel: [    0.949019] setup_mtd_parts: name misc1 configured size 0x00800000 offset 0xBF00000
May  5 15:05:22 kernel: [    0.956964] Creating 11 MTD partitions on "Micron MT29F2G01AB":
May  5 15:05:22 kernel: [    0.962902] 0x000000100000-0x000006000000 : "rootfs"
May  5 15:05:22 kernel: [    0.999703] 0x000006000000-0x00000bf00000 : "rootfs_update"
May  5 15:05:22 kernel: [    1.036983] 0x00000f700000-0x00000ff00000 : "data"
May  5 15:05:22 kernel: [    1.044796] 0x000000000000-0x000000100000 : "nvram"
May  5 15:05:22 kernel: [    1.050355] 0x000000100000-0x000006000000 : "image"
May  5 15:05:22 kernel: [    1.086935] 0x000006000000-0x00000bf00000 : "image_update"
May  5 15:05:22 kernel: [    1.124083] 0x000000000000-0x000010000000 : "dummy1"
May  5 15:05:22 kernel: [    1.213703] 0x000000000000-0x000010000000 : "dummy2"
May  5 15:05:22 kernel: [    1.303320] 0x00000f600000-0x00000f700000 : "misc3"
May  5 15:05:22 kernel: [    1.308896] 0x00000c700000-0x00000f600000 : "misc2"
May  5 15:05:22 kernel: [    1.329626] 0x00000bf00000-0x00000c700000 : "misc1"
May  5 15:05:22 kernel: [    1.387760] brcmboard registered
May  5 15:05:22 kernel: [    1.390989] brcmboard: brcm_board_init entry
May  5 15:05:22 kernel: [    1.395279] print_rst_status: Last RESET due to SW reset
May  5 15:05:22 kernel: [    1.400598] print_rst_status: RESET reason: 0x80000000
May  5 15:05:22 kernel: [    1.405795] DYING GASP IRQ Initialized and Enabled
May  5 15:05:22 kernel: [    1.410613] SES: LED GPIO 0x8018 is enabled
May  5 15:05:22 kernel: [    1.414818] map_hw_timer_interrupt,130: interrupt_id 22
May  5 15:05:22 kernel: [    1.420055] map_hw_timer_interrupt,130: interrupt_id 23
[... cut ...]
May  5 15:09:11 rc_service: wanduck 1562:notify_rc restart_wan_if 0
May  5 15:09:15 rc_service: wanduck 1562:notify_rc restart_wan_if 1
May  5 15:09:22 BWDPI: force to flush flowcache entries
May  5 15:09:22 BWDPI: rollback fc
May  5 15:09:33 rc_service: httpd 1672:notify_rc start_dsl_autodet
May  5 15:09:48 rc_service: wanduck 1562:notify_rc restart_wan_if 1
May  5 15:09:52 rc_service: wanduck 1562:notify_rc restart_wan_if 0
May  5 15:10:55 rc_service: wanduck 1562:notify_rc restart_wan_if 0
May  5 15:10:59 rc_service: wanduck 1562:notify_rc restart_wan_if 1

DJE007, 2022-07-02:
You wish any commands run and output from ssh console?

DJE007, 2022-07-04:
Same with beta2

DJE007, 2022-07-04:
[image]

DJE007, 2022-07-04:
Why is DHCP on both eth4 and Wan0?
Should only be on Wan0 right

DJE007, 2022-07-04:
Ethernet dev's

optus@DSL-AX5400-8F88:/tmp/home/root# ps | grep dhcp
 3627 optus     3424 S    /sbin/udhcpc -i eth4 -p /var/run/udhcpc1.pid -s /tmp/udhcpc_wan -t1 -T5 -A0 -O33 -O249 -x 125 00000de9210106374331304339020c3743313043393336384638
 3641 optus     3424 S    /sbin/udhcpc -i wan0 -p /var/run/udhcpc0.pid -s /tmp/udhcpc_wan -t1 -T5 -A0 -O33 -O249 -x 125 00000de9210106374331304339020c3743313043393336384638
 3773 optus     3424 S    grep dhcp

optus@DSL-AX5400-8F88:/tmp/home/root# ip a
1: lo: <LOOPBACK,MULTICAST,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 brd 127.255.255.255 scope host lo
       valid_lft forever preferred_lft forever
    inet 127.0.1.1/8 brd 127.255.255.255 scope host secondary lo:0
       valid_lft forever preferred_lft forever
2: ifb0: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether 3e:b9:6f:f2:77:8b brd ff:ff:ff:ff:ff:ff
3: ifb1: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether ae:78:42:16:81:61 brd ff:ff:ff:ff:ff:ff
4: imq0: <NOARP> mtu 16000 qdisc noop state DOWN group default qlen 11000
    link/void
5: imq1: <NOARP> mtu 16000 qdisc noop state DOWN group default qlen 11000
    link/void
6: imq2: <NOARP> mtu 16000 qdisc noop state DOWN group default qlen 11000
    link/void
7: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default
    link/sit 0.0.0.0 brd 0.0.0.0
8: ip6tnl0@NONE: <NOARP> mtu 1452 qdisc noop state DOWN group default
    link/tunnel6 :: brd ::
9: ip6gre0@NONE: <NOARP> mtu 1448 qdisc noop state DOWN group default
    link/gre6 :: brd ::
10: archer: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
11: dsl0: <> mtu 0 qdisc noop state DOWN group default
    link/[29]
12: bcmsw: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 7c:10:c9:36:8f:88 brd ff:ff:ff:ff:ff:ff
13: eth0: <NO-CARRIER,BROADCAST,MULTICAST,ALLMULTI,UP> mtu 1500 qdisc pfifo_fast master br0 state DOWN group default qlen 1000
    link/ether 7c:10:c9:36:8f:88 brd ff:ff:ff:ff:ff:ff
14: eth1: <NO-CARRIER,BROADCAST,MULTICAST,ALLMULTI,UP> mtu 1500 qdisc pfifo_fast master br0 state DOWN group default qlen 1000
    link/ether 7c:10:c9:36:8f:88 brd ff:ff:ff:ff:ff:ff
15: eth2: <NO-CARRIER,BROADCAST,MULTICAST,ALLMULTI,UP> mtu 1500 qdisc pfifo_fast master br0 state DOWN group default qlen 1000
    link/ether 7c:10:c9:36:8f:88 brd ff:ff:ff:ff:ff:ff
16: eth3: <NO-CARRIER,BROADCAST,MULTICAST,ALLMULTI,UP> mtu 1500 qdisc pfifo_fast master br0 state DOWN group default qlen 1000
    link/ether 7c:10:c9:36:8f:88 brd ff:ff:ff:ff:ff:ff
17: eth4: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether 7c:10:c9:36:8f:8c brd ff:ff:ff:ff:ff:ff
18: dpsta: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
19: eth5: <BROADCAST,MULTICAST,ALLMULTI,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast master br0 state UP group default qlen 1000
    link/ether 7c:10:c9:36:8f:88 brd ff:ff:ff:ff:ff:ff
20: eth6: <BROADCAST,MULTICAST,ALLMULTI,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast master br0 state UP group default qlen 1000
    link/ether 7c:10:c9:36:8f:8c brd ff:ff:ff:ff:ff:ff
21: br101: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UNKNOWN group default
    link/ether 96:e1:0f:d7:6f:42 brd ff:ff:ff:ff:ff:ff
    inet 169.254.10.99/16 brd 169.254.255.255 scope global br101
       valid_lft forever preferred_lft forever
22: br0: <BROADCAST,MULTICAST,ALLMULTI,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default
    link/ether 7c:10:c9:36:8f:88 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.1/24 brd 192.168.0.255 scope global br0
       valid_lft forever preferred_lft forever
23: ptm0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 7c:10:c9:36:8f:88 brd ff:ff:ff:ff:ff:ff
24: wan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default
    link/ether 7c:10:c9:36:8f:88 brd ff:ff:ff:ff:ff:ff

optus@DSL-AX5400-8F88:/tmp/home/root# ip r
127.0.0.0/8 dev lo scope link
169.254.0.0/16 dev br101 proto kernel scope link src 169.254.10.99
192.168.0.0/24 dev br0 proto kernel scope link src 192.168.0.1
239.0.0.0/8 dev br0 scope link
optus@DSL-AX5400-8F88:/tmp/home/root#

gnuton, 2022-07-05:
ok you have an optus variant. -_-
You have two DHCP because you have dualwan enabled one on wan0 and the other one on eth4.

Can you plese reset the factory the router and just add VDSL config and no other ones?

DJE007, 2022-07-05:
> ok you have an optus variant. -_- You have two DHCP because you have dualwan enabled one on wan0 and the other one on eth4.
> 
> Can you plese reset the factory the router and just add VDSL config and no other ones and let's see if it works?

That's what I have done to start with, beta 2 all I had done is turn on sshd after reset to factory.

DJE007, 2022-07-06:
> do you have dual wan enabled too?

enabled yes, use no I only use DSL link

DJE007, 2022-07-29:
Fixed in Asus code base for newer firmware

Version 3.0.0.4.386.49608
2022/07/28 55.83 MBytes
ASUS DSL-AX82U Firmware version 3.0.0.4.386.49608
Security
- Fixed string format stacks vulnerability
- Fixed cross-site-scripting vulnerability
- Fixed informational vulnerability. Thanks to Howard McGreehan.
- Fixed SQL injection vulnerability
- Fixed json file traversal vulnerability
- Fixed plc/port file traversal vulnerability
- Fixed stack overflow vulnerability. Thanks to HP of Cyber Kunlun Lab
- Fixed authenticated stored XSS vulnerability. Thanks to Luke Walker – SmartDCC
- Fixed cfgserver heap overflow vulnerability
- Fixed cfgserver denial of service vulnerability. Thanks to TianHe from BeFun Cyber Security Lab.
- Fixed OpenSSL CVE-2022-0778
- Fixed CVE-2021-34174
- Added more security measures to block malware.
- Fixed Stored XSS vulnerability. Thanks to Milan Kyselica of IstroSec.
- Fixed CVE-2022-23970, CVE-2022-23971, CVE-2022-23972, CVE-2022-23973, CVE-2022-25595, CVE-2022-25596, CVE-2022-25597, CVE-2022-26376

Bug fixes
- Fixed AiMesh guest network issues.
- Fixed DDNS issues where the WAN IP is IPv6
- Fixed UI bugs in Administration --> feedback.
- Fixed time zone error.
- Fixed Aura RGB issue.
- Fixed Internet connection issue for specific ISPs.

New features
- Added 3rd party DNS server list in WAN --> DNS to help users enhance the connection security.
- Supported Safe Browsing in the router app to filter explicit content from search results. You can set it in the router app --> Devices or Family.

gnuton, 2022-08-11:
Please test this image and let me know if it works.
Note: I have not tested it before hand.. so testing is at your own risk.
I cannot test it before next week. 

https://output.circle-artifacts.com/output/job/7c74daa1-42e7-4eb9-89cb-9f9638a84a6c/artifacts/0/workspace/release/DSL-AX82U_386.8_beta2_puresqubi.w

gnuton, 2022-08-12:
@PreoPhil please follow this issue

gnuton, 2022-08-15:
@DJE007  I tested this on my DSL-AX82U and it shoudl be safe now.

Please. flash this and let me know if it works now

https://output.circle-artifacts.com/output/job/242d383e-6fdf-45b9-bd57-97c46ae435c3/artifacts/0/workspace/release/DSL-AX82U_386.8_0_puresqubi.w

DJE007, 2022-08-16:
No sorry didn't work

gnuton, 2022-08-18:
@DJE007 did you try the latest stock firmware? does it work?

DJE007, 2022-08-18:
I tried the one you posted in #204

FreoPhil, 2022-10-09:
Hi, I today tried the 386.08_0-gnuton1 release for DSL-AX82U. I continue to have the same issue in that DSL comes up, but never gets an IP from ISP provider. ISP provider is Aussie Broadband. This appears to be the same issue reported by DJE007 and the same one I reported on 386.07 and the next beta, which I both tried repeatedly. I have rolled back to 386.5_2 so I have internet.

viTRifY13, 2022-10-11:
I also agree with FreoPhil as I have the DSL-AX5400 and I'm with Optus.  386.5_2 is the only version that works / authenticates the DSL connection.  386.08_0 fails to authenticate the DSL.  I flashed both firmwares, set the logging level to debug, cleared the log, unplugged the RJ12 data cable, waited a few seconds, plugged it back in, waited a couple of minutes to authenticate the DSL and then compared the logs between the 2 versions.  Comparison is below.  386.5_2 works, but 386.08 doesn't

Highlighted in red are the differences between the 2 logs

[image]

DJE007, 2022-10-18:
386.08_0-gnuton1 works now

FreoPhil, 2022-10-18:
I’m really sorry, but I have tried your latest version twice now and it still doesn’t work. Exactly the same issue with DHCP.

Have gone back to version 386.5_2 again.

Regards

Philip White
m - +61 (0) 409 044 509

Closed #204<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fgnuton%2Fasuswrt-merlin.ng%2Fissues%2F204&data=05%7C01%7C%7C761959b16cff46f5f16808dab0da3b90%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638016748596471566%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=jXZzgR5JkVGzbQVisZEwkxYGYAc%2FVSAYif5%2FFb6B0lA%3D&reserved=0> as completed.

DJE007, 2022-10-18:
> I’m really sorry, but I have tried your latest version twice now and it still doesn’t work. Exactly the same issue with DHCP. Have gone back to version 386.5_2 again. 

Are you with Optus?

Looks like you are not the only one
https://forums.whirlpool.net.au/thread/32k1vyn9?p=-1#bottom

FreoPhil, 2022-10-18:
No, Aussie Broadband.

Regards

Philip White
m - +61 (0) 409 044 509

https://forums.whirlpool.net.au/thread/32k1vyn9?p=-1#bottom<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fforums.whirlpool.net.au%2Fthread%2F32k1vyn9%3Fp%3D-1%23bottom&data=05%7C01%7C%7Ce325060b64de46d831bc08dab14eed6d%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638017249802853280%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=%2Fso5MnMJxmPke9JSh90iEbv9WJcrkcjMH0y9qeV8Mks%3D&reserved=0>

FreoPhil, 2022-11-03:
I still don't have a resolution to this issue, yet the issue is closed. Can you help or direct me to where I might find the solution or thread discussing the progress of this issue?

gnuton, 2022-11-05:
@FreoPhil I am getting new 388 GPLs so I would try to get some alpha for you to test.

FreoPhil, 2022-11-07:
As background, the last version that does not display this issue, is 386.5_2-gnuton0. That is what I keep having to roll back too, to get a connection. Hope this helps.

viTRifY13, 2022-11-07:
> As background, the last version that does not display this issue, is 386.5_2-gnuton0. That is what I keep having to roll back too, to get a connection. Hope this helps.

Same here as well. 386.5_2 is the last one that I get a connection with also

FreoPhil, 2022-11-19:
Any progress?

avapug, 2022-12-05:
Still not working on 388.1_0-gnuton0_beta1

avapug, 2022-12-23:
AussieBB is working for me on 388.1_0-gnuton0_beta2
Thing's I changed were setting DHCP query frequency to Normal and paying for a static ip.

FreoPhil, 2022-12-26:
I today uploaded 388.1_0-gnuton0_beta2 and it successfully connected to Aussie BB. So far so good

gnuton, 2022-12-27:
Awesome! I knew it.
I have enanabled a new IPV6 feature which was not fully complete in the previous GPL and  I do have cleaned up a few other provider settings. 

Glad that worked out!

gnuton, 2022-12-27:
@avapug what s your provider?

avapug, 2022-12-27:
Aussie Broadband

FreoPhil, 2022-12-27:
Mine too, Aussie Broadband

viTRifY13, 2022-12-28:
Successfully connected with Optus.  GREAT WORK @gnuton !!!

Thanks!