gnuton/asuswrt-merlin.ng #205: Missing picture in RT-AX82U

gnuton, 2022-06-30:
Have had the current 386.07_0-gnuton0_beta1 running on my RT-AX82U , working great by the way. The only small problem, I have, is that the router picture is not displaying, node pictures display without any issue, and phone app also displays ALL router pictures without any issue. Have attached a scrren dump of the error from gui, using the F12 develpment console, from Chrome.

Note I have tried multiple full resets and different broswers, clearing cache.. No ad blocking software.

Current GUI URL:

https://nw-dlcdnet.asus.com/plugin/productIcons/RT-AX82U_CB_0.png - does not work - 404 error not found. The requested URL was not found on this server.

Change the above GUI URL, to this link within the browser:

https://nw-dlcdnet.asus.com/plugin/productIcons/RT-AX82U.png - works, an image is found

this comes from https://www.snbforums.com/threads/fork-dsl-testers-gnutons-alpha-beta-merlin-builds.71150/page-8#post-772764

gnuton, 2022-07-03:
Confirmed. No idea yet what's to he problem

neil1962, 2022-07-17:
FYI, I also tried the current ASUS stable release for the RT-AX82U, and this problem also exists.

neil1962, 2022-12-15:
The problem is now fixed, picture is now displaying, ASUS must of fixed the problem within their own server

gnuton, 2022-12-16:
awesome thanks I will fix this!

closed 2022-12-16 by gnuton