gnuton/asuswrt-merlin.ng #207: Minor bug (?) shows in log.

GaselK, 2022-07-03:
**Router Model Affected**
Models: TUF AX-5400

**Firmware Version Affected**
386.07_0-gnuton0_beta2

**Describe the bug**
Log entries. 
"
Jul  3 10:07:53 kernel: nf_conntrack: expectation table full
Jul  3 10:07:53 kernel: nf_conntrack: expectation table full
Jul  3 10:07:53 kernel: nf_conntrack: expectation table full
Jul  3 10:07:53 kernel: nf_conntrack: expectation table full
Jul  3 10:07:53 kernel: nf_conntrack: expectation table full
Jul  3 10:07:53 kernel: nf_conntrack: expectation table full
Jul  3 10:07:56 kernel: nf_conntrack: expectation table full
Jul  3 10:07:56 kernel: nf_conntrack: expectation table full
Jul  3 10:07:56 kernel: nf_conntrack: expectation table full
Jul  3 10:07:56 kernel: nf_conntrack: expectation table full
Jul  3 10:08:00 kernel: nf_conntrack: expectation table full
Jul  3 10:08:00 kernel: nf_conntrack: expectation table full
Jul  3 10:08:00 kernel: nf_conntrack: expectation table full
Jul  3 10:08:00 kernel: nf_conntrack: expectation table full
Jul  3 10:08:00 kernel: nf_conntrack: expectation table full
Jul  3 10:08:00 kernel: nf_conntrack: expectation table full
Jul  3 10:08:00 kernel: nf_conntrack: expectation table full
Jul  3 10:08:00 kernel: nf_conntrack: expectation table full
Jul  3 10:08:00 kernel: nf_conntrack: expectation table full
Jul  3 10:08:00 kernel: nf_conntrack: expectation table full
Jul  3 10:08:05 kernel: nf_conntrack: expectation table full
Jul  3 10:08:06 kernel: nf_conntrack: expectation table full
Jul  3 10:08:06 kernel: nf_conntrack: expectation table full
Jul  3 10:08:06 kernel: nf_conntrack: expectation table full
Jul  3 10:08:06 kernel: nf_conntrack: expectation table full
Jul  3 10:08:06 kernel: nf_conntrack: expectation table full
Jul  3 10:08:07 kernel: nf_conntrack: expectation table full
Jul  3 10:08:07 kernel: nf_conntrack: expectation table full
"

**To Reproduce**
Steps to reproduce the behavior:
1. Go to System Log

**Expected behavior**
I can't tell if this a bug (since it existed and in previous betas) or normal behavior.
Router works perfectly stable nevertheless.

gnuton, 2022-07-03:
Hi do you have skynet or any other script?
There is a thread about this issue: https://www.snbforums.com/threads/nf_conntrack-expectation-table-full-and-other-log-oddities.55415/page-4

gnuton, 2022-07-03:
I cannot reproduce this, which .means reset factory to fix it this

GaselK, 2022-07-04:
Factory reset after flash. Settings by hand. JFFS not enabled. (no scripts).

gnuton, 2022-07-05:
do you still see the problem if disable parental control? (maybe the culprit) and/or DoT?

GaselK, 2022-07-06:
Yes. Tried DoT disabled only, Parental disabled only and both disabled.

gnuton, 2022-07-06:
FYI I hav tested this and I am not getting it.. so not a blocker for the stable release as for now.
You can mitigate this with some script wihch are in the link I sent you. you can tell the kernel to allocate larger table

GaselK, 2022-07-06:
log entry hasn't show up in log the laste 24hrs. No way to reproduce so i will mark this thread closed.

closed 2022-07-06 by GaselK