gnuton/asuswrt-merlin.ng #212: TUF AX5400 icons missing

GaselK, 2022-07-09:
**Router Model Affected**
Models: TUF AX 5400

**Firmware Version Affected**
TUF-AX5400_386.07_0-gnuton0_beta2_cferom_puresqubi

**Describe the bug**
Icons missing from WiFi Radar

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Wireless
2. Click on WiFi Radar
3. See error

**Screenshots**
[image]

gnuton, 2022-07-09:
thanks I fixed this the other night! 
[image]

closed 2022-07-09 by gnuton

d3v3l15h, 2022-11-13:
Hi,

First, thanks for these firmwares.

Just want to say I have this very same issue on XT8 with build 386.07_2-gnuton1. Not sure if the fix has been deployed though.

EM

gnuton, 2022-11-14:
@d3v3l15h I am preparing the 388 builds I will keep an aye on the new ones. 
In theory this should not be the case for XT8, usually missing icons issues are found only in the first images I release for a model.

d3v3l15h, 2022-11-14:
OK Thanks.

[image]

EM