gnuton/asuswrt-merlin.ng #214: RT-AX82U. Question mark shown in the page

gnuton, 2022-07-15:
**Router Model Affected**
Models: RT-Ax82u
[image]

gnuton, 2022-07-15:
the offending code is this
 if(nt_center_support)
                banner_code +='<td width="30" id="notification_status_td"><div><div id="noti_event_count" style="display:none"><div id="noti_event_num"></div></div><div id="notification_status"></div></div><div id="notification_desc"></div></td>\n';
        else
                banner_code +='<td width="30" id="notification_status1" class="notificationOn"><div id="notification_status" class="notificationOn"></div><div id="notification_desc" class=""></div></td>\n';

nt_center_support is 0 so the second banner_code is appened.

 var nt_center_support = (isSupport("nt_center") && isSupport("nt_center_ui"))

gnuton, 2022-07-15:
[image]
BRoken

gnuton, 2022-07-15:
[image]
the working one on DSL-AX82U

gnuton, 2022-07-16:
fixed 
[image]

neil1962, 2022-07-17:
Tahnks will try on your next firmware release.

gnuton, 2022-07-27:
this fixed is part of beta4

neil1962, 2022-07-27:
I can confirm problem is fixed,  thank you

closed 2022-07-27 by gnuton

neil1962, 2022-07-27:
Thanks

Sent From My Mobile Device

Closed #214<https://github.com/gnuton/asuswrt-merlin.ng/issues/214> as completed.