gnuton/asuswrt-merlin.ng #215: TUF AX5400 gui font mismatch

GaselK, 2022-07-15:
**Router Model Affected**
Models: TUF AX5400

**Firmware Version Affected**
386.07_0-gnuton0_beta3

**Describe the bug**
In WAN screen font is mismatch from main font used and a strange icon appears over qos settings icon

**To Reproduce**
Steps to reproduce the behavior:
1. Go to WAN

**Screenshots**
If applicable, add screenshots to help explain your problem or remove this section
[image]

gnuton, 2022-07-15:
this glitch as this https://github.com/gnuton/asuswrt-merlin.ng/issues/214

GaselK, 2022-07-27:
Fixed in 386.07_2-gnuton0_beta4

closed 2022-07-27 by GaselK