gnuton/asuswrt-merlin.ng #219: Flexos cannot run

micron888, 2022-07-26:
Tuf ax5400 beta 3
No test on stable build

micron888, 2022-07-26:
Amtm
Install flex os
Under sys log
Under qos. Cannot see flexqos in the menu tab
 
Jul 25 08:04:21 custom_script: Running /jffs/scripts/service-event-end (args: restart qos)
Jul 25 08:04:21 custom_script: Running /jffs/scripts/firewall-start (args: eth4)
Jul 25 08:04:21 custom_script: Running /jffs/scripts/service-event-end (args: restart firewall)
Jul 25 08:04:22 FlexQoS: /jffs/addons/flexqos/flexqos.sh (pid=1883) called in unattended mode with 1 args: -start
Jul 25 08:04:22 FlexQoS: Applying iptables static rules
Jul 25 08:04:22 FlexQoS: Applying iptables custom rules
Jul 25 08:04:23 FlexQoS: Flushing conntrack table
Jul 25 08:04:25 FlexQoS: Applying AppDB rules and TC rates

gnuton, 2022-07-28:
the error is "  Cannot see flexqos in the menu tab"
that's not part of the firmware but you have to install it
You may wanna check https://github.com/dave14305/FlexQoS

gnuton, 2022-07-28:
As for now there is nothing to be fixed and ithis task has been marked as "tech support" ticket.

dave14305, 2022-07-30:
@gnuton would you update this for tuf5400 to use eth4 as wan_iface?

https://github.com/gnuton/asuswrt-merlin.ng/blob/790b5524e24f7eb9985efc60d2dc20a1620baab2/release/src/router/httpd/data_arrays.c#L683-L689

reaps101, 2022-09-05:
will this get changed or is this something we can change by ourself if pointed in the direction thanks

gnuton, 2022-09-05:
@reaps101 I fixed this when i was on holiday but i didn't have any router to test and then I forgot it. This should theoretically fix the issue "forever" https://github.com/gnuton/asuswrt-merlin.ng/commit/0eeb7ec17e2b0dd551d6cecae303f20231d09a88
I will test this now that I am at home and i will merge if it works fine across all models
stay tuned

gnuton, 2022-10-13:
Experimental images with the fix:

- https://output.circle-artifacts.com/output/job/fa1f19a1-45ae-4394-9361-bcbf61255f21/artifacts/0/workspace/release/TUF-AX5400_386.7_2_cferom_puresqubi.w

- https://output.circle-artifacts.com/output/job/40e48ee9-74ff-4805-ab09-0c11680bf162/artifacts/0/workspace/release/TUF-AX3000_386.7_2_cferom_puresqubi.w

reaps101, 2022-10-14:
hi @gnuton so flexqos seems to be running great now everything seems to be working what i tested so far with this patch, only thing left would be for @dave14305 to update flexqos.sh so we dont have to change code manually everytime "replace QoS_Stats.asp with AdaptiveQoS_WebHistory.asp"

micron888, 2022-10-14:
So far so good.
[image]